miguelgrinberg/microdot #1: Have a little trouble to understand source

zhazhalaila, 2019-07-01:


miguelgrinberg, 2019-07-01:
The response class is defined in microdot.py.

zhazhalaila, 2019-07-01:
> The response class is defined in microdot.py.

Yeah. I get this. 
In the microdot.py, I notice two line
```
 361               if res is None:
 362                   res = f(req, **req.url_args)
```
When I run gpio.py, I access `http://0.0.0.0:5000`. It will execute this two line. After than I get a `response class`. But what to make me confused is that I don't create `response class` and access it as parameter.

So where is the way to create `response class`.

In my understanding, I think i should write this code

microdot.py
```
 361               if res is None:
 362                   res = Response()
 363                   res = f(req, res, **req.url_args)
```

gpio.py
```
@app.route('/', methods=['GET', 'POST'])
def index(request, response):
    pass
```

miguelgrinberg, 2019-07-01:
That's not how this framework works. The response object is created after the handler returns: See https://github.com/miguelgrinberg/microdot/blob/master/microdot/microdot.py#L361-L366.

zhazhalaila, 2019-07-01:
> That's not how this framework works. The response object is created after the handler returns: See https://github.com/miguelgrinberg/microdot/blob/master/microdot/microdot.py#L361-L366.

But I don't wirte `@before_request` or `@after_request`
In gpio.py, `response class` from where?
Why I can write `response.set_cookie('form', form_cookie)` in gpio.py and no need to import microdot.Response?

miguelgrinberg, 2019-07-01:
I don't understand what you are finding confusing. The `redirect()` and `send_file()` functions both return ` Response` object already created. You can invoke any methods of this object without having to import the object itself. The class needs to be available only if you need to create a new object, but in this situation the object is created by the framework.

zhazhalaila, 2019-07-01:
> I don't understand what you are finding confusing. The `redirect()` and `send_file()` functions both return ` Response` object already created. You can invoke any methods of this object without [...]

I don't focus on `redirect()` and `send_file()`, I'm so silly.  So grateful for you!

closed 2019-07-01 by zhazhalaila