miguelgrinberg/microdot #2: upip install fails

xorbit, 2020-02-18:
Hi Miguel,

Even though Microdot seems to be in the Pypi repo, I cannot install it.
I'm using "MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32"

```python
>>> upip.install('microdot')
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Error installing 'microdot': , packages may be partially installed
>>> import microdot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'microdot'
>>> 
```

Any ideas?

miguelgrinberg, 2020-02-18:
Unfortunately upip isn't giving an error message, note the "," between the colon and the word "packages". It's supposed to show an error message there and it isn't, so I do not know. I can confirm that I get the same error on the ESP8266, and also on the MacOS release.

The fact it happens on several platforms including a desktop one suggests it is not memory related. I'll have to investigate this more to know what's going on.

miguelgrinberg, 2020-02-18:
Can you try again? The problem was caused by having wheel package along with the source distribution uploaded to PyPI. That confuses upip. I have now removed all the wheel packages.

On the Mac MicroPython I can get it installed just fine. On the ESP8266 it fails with a different error, which could be memory. I don't have an ESP32 at hand right now, so can you please test it on your side and report?

xorbit, 2020-02-18:
It works now on ESP32, thanks for the quick fix!
And for providing a good alternative to Picoweb, which is now broken.

miguelgrinberg, 2020-02-18:
Picoweb has been broken for a very long time. That is one of the reasons I created microdot! Thanks.

closed 2020-02-18 by miguelgrinberg

maho, 2021-02-21:
I don't know if I should reopen this ticket, or create new - but I got this error right now, and I think it is not related to memory, because I'm using unix port. 

However, micropython at esp32 gives the very same error message:

```
maho@ustat: ~/workspace/microdot/examples master $ micropython -m upip install -p lib microdot                              [20:06:54]
Installing to: lib/
Warning: micropython.org SSL certificate is not validated
Error installing 'microdot': , packages may be partially installed
maho@ustat: ~/workspace/microdot/examples master $  
```

miguelgrinberg, 2021-02-21:
@maho: I forgot that with upip I have to prevent a wheel package from being uploaded. Try again now, I have removed the wheel files.

maho, 2021-02-22:
works, however I will switch to copying files manually :). At least for `microdot`. 

Eg. I got this error now (not sure if it's ok or not, rather not):

```
maho@ustat % micropython -m upip install -p lib microdot-asyncio                                                                                                          /tmp
Installing to: lib/
Warning: micropython.org SSL certificate is not validated
Installing microdot-asyncio 0.3.1 from https://files.pythonhosted.org/packages/69/54/208c8c5778bfc356b3da984feb332ec66f78ea1c751080ebdd8cfb6fc194/microdot-asyncio-0.3.1.tar.gz
Installing microdot 0.3.1 from https://files.pythonhosted.org/packages/f1/a4/578d00712dece1d103c01be91488cf2290adceaf0a086ac36626b230ef3d/microdot-0.3.1.tar.gz
Error installing '': Package not found, packages may be partially installed
m
```

joesilva01862, 2021-09-29:
Hi Miguel. I just saw this page now.

My attempt to install indicates that it was successful, like this:
>>> upip.install('microdot')
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing microdot 0.7.2 from https://files.pythonhosted.org/packages/31/ed/f080294e631725f84acd0317e162fa81f6de00c7590e9a7ab7b89c2262e3/microdot-0.7.2.tar.gz

But, just by looking under the /lib folder on my ESP32 device, there is no microdot folder, and some individual files are actually installed under the lib folder.

Additionally, when I try to import, I get this message:
>>> from microdot import Microdot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'microdot'

Would you help? Thanks.

miguelgrinberg, 2021-09-29:
@joesilva01862 When you install microdot you end up with two files, microdot.py and microdot_asyncio.py. There is no microdot folder. So what you are describing seems correct, as long as your /lib directory is in the import path in your micropython installation.

In any case, what I recommend is that you put these two files in your main project directory instead of in /lib. And see if that works. I assume that will work, so now that you have everything working, you can relocate the files to wherever you like, and then adjust the import path accordingly. For example, if you move the files to /lib, you will then import as follows:

    from lib.microdot import Microdot

Note that if you use the asyncio version of Microdot, after moving these two files to a directory you will need to adjust the imports in microdot_asyncio.py so that this file can find microdot.py in the new location.

joesilva01862, 2021-09-29:
Miguel, thank you for your reply.
With your help, I got a little farther, but still no success.
When run this test program below, I get the message below. Line 824 from microdot.py is this one here:
res = f(req, **req.url_args)

Any idea how to move beyond this error? Thanks.
Joe

Program:

from src.microdot import Microdot
import network

#-------------------------------------------
# configure the device as an Access Point
#-------------------------------------------
ssid = 'MicroPython-Dongle'
password = '12345678'

ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(essid=ssid, password=password, authmode=3)

# make the access point active
while ap.active() == False:
  pass

config_info = ap.ifconfig()
print(config_info)

app = Microdot()

@app.route('/')
def index():
    return 'Hello, world!'

app.run(port=5000, debug=True)

('192.168.4.1', '255.255.255.0', '192.168.4.1', '0.0.0.0')
Starting threaded server on 0.0.0.0:5000...

Log message when I try to hit endpoint "/":

Traceback (most recent call last):
  File "/lib/src/microdot.py", line 824, in dispatch_request
TypeError: function takes 0 positional arguments but 1 were given
GET / 500
GET /favicon.ico 404

miguelgrinberg, 2021-09-29:
@joesilva01862 Are you used to Flask? A design difference in Microdot is that there is no app/request contexts. The request object is passed as an argument to your view functions. Your `index()` function needs to accept `request` as an argument.

joesilva01862, 2021-09-29:
@miguelgrinberg It's working now. My appreciation for your great work.

joesilva01862, 2021-11-03:
Hi Miguel. I've been making great progress with my side project on ESP32. The question now is how would I go about creating an SSL-like request/response with Microdot? 
The data coming from the client and the server's response would immensely benefit from encryption.

Thanks.
Joe

miguelgrinberg, 2021-11-04:
@joesilva01862 I have not implemented SSL encryption for the web server. I think this would just require some changes in the `run()` method, but haven't investigated fully.