miguelgrinberg/microdot #3: send_file fails with larger files

xorbit, 2020-02-18:
`send_file` is great but it seems to have an issue with larger files.

When trying to send an 18k JPG, I get this:

```python
Unhandled exception in thread started by <closure>
Traceback (most recent call last):
  File "/lib/microdot.py", line 24, in run
  File "/lib/microdot.py", line 390, in dispatch_request
  File "/lib/microdot.py", line 210, in write
OSError: [Errno 104] ECONNRESET
```

Looking at the code, I think a likely problem is that it reads the whole file into memory before it sends it.  Is there a way to do this differently, more like a stream architecture, so it doesn't need so much memory?

miguelgrinberg, 2020-02-18:
Yes, large files would not work with the current implementation.

The `Response.write()` method assumes `self.body` contains the bytes, all in memory. We can accept a file-like object as body, and then the write can be done in small chunks. Would that work?

xorbit, 2020-02-18:
It probably would work, if I knew how exactly. :)

I mostly use MicroPython in support of my hardware (wESP32) to support my customers, but I'm by no means an expert.

Could you provide a simple example on how to implement an endpoint to serve files that way?

miguelgrinberg, 2020-02-19:
Give the microdot.py version in the master branch a try. I have changed it to send file downloads in blocks of 1024 bytes now.

closed 2020-02-19 by miguelgrinberg

xorbit, 2020-02-19:
Hi Miguel,

Tested it, but it's kind of odd.  It loads the image (29K now), but it throws errors all the time as well:

```python
Unhandled exception in thread started by <closure>
Traceback (most recent call last):
  File "/lib/microdot.py", line 24, in run
  File "/lib/microdot.py", line 399, in dispatch_request
  File "/lib/microdot.py", line 211, in write
OSError: [Errno 104] ECONNRESET
```

xorbit, 2020-02-19:
Another thing.  I have a test app that keeps making AJAX calls, and after running a while, this happens:

```python
Traceback (most recent call last):
  File "main.py", line 57, in <module>
  File "/lib/microdot.py", line 343, in run
  File "/lib/microdot.py", line 26, in create_thread
OSError: can't create thread
MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```

That's not good.
Maybe would be a good idea to get an ESP32 so you can test in memory constrained environments?

miguelgrinberg, 2020-02-20:
I'll test with images to see if I get this problem here.

I'll also look into the thread issue. I expected threads would be cleaned up when they end, but maybe I'm wrong there. Memory problems can also be caused by your application leaking resources I guess, depending on what you do on your web handlers.

xorbit, 2020-02-20:
My handlers are very simple:

```python
# Home page
@app.route("/")
def index(request):
  return send_file('static/index.html')

# Logo image
@app.route("/static/wesp32-logo.png")
def logo(request):
  return send_file('static/wesp32-logo.png')

# Update light
@app.route("/light")
def light(request):
  if 'level' in request.args:
    ledpwm.duty(int(request.args['level']))
  return str(ledpwm.duty())

# Memory info
@app.route("/memory")
def memory(request):
  if 'gc' in request.args or 'collect' in request.args:
    print ("Garbage collect")
    gc.collect()
  return str(gc.mem_free())
```

`/light` is the one that's called continuously by the page.

xorbit, 2020-03-03:
Any update on this?  In lieu of that, how does one use the asyncio version?

commit 3e29af57 (2020-02-19): Support large downloads in send_file (fixes #3)