miguelgrinberg/microdot #5: send_file fails with larger files in asyncio version

xorbit, 2020-03-05:
Was the issue in #3 also fixed in the asyncio version?

When trying to send a 30 kB file this happens:

```python
Traceback (most recent call last):
  File "main.py", line 57, in <module>
  File "microdot_asyncio.py", line 100, in run
  File "/lib/uasyncio/core.py", line 180, in run_until_complete
  File "/lib/uasyncio/core.py", line 109, in run_forever
  File "microdot_asyncio.py", line 94, in serve
  File "microdot_asyncio.py", line 145, in dispatch_request
  File "microdot_asyncio.py", line 72, in write
  File "/lib/uasyncio/__init__.py", line 169, in awrite
OSError: [Errno 104] ECONNRESET
```
The picture was only partially loaded and the web server dead after this (back to Python prompt).

miguelgrinberg, 2020-03-05:
Both versions received the same fix, but none of them made it to PyPI yet, you have to install from the master branch on this repo.

xorbit, 2020-03-05:
Well, since I couldn't install from PyPi (due to #4), I did get it from master here and that's the version that exhibited the error.

xorbit, 2020-03-05:
I'm confused.  You fixed it so the non-async version works with this 30 kB image, but the async doesn't and gives the above error.

miguelgrinberg, 2020-03-05:
This is the same error you have reported on the non-asyncio version, which makes sense because both have received the same change.

miguelgrinberg, 2020-03-06:
Yes, on the asyncio version there is just one thread. If there is an error that is not handled the server would stop running. Regardless of the solution to these errors, I will be making sure I catch any possible errors to prevent the server from ending.

xorbit, 2020-03-06:
Thanks Miguel, I appreciate your efforts!
At the moment I have the issue that I can't really recommend any MicroPython web server solution to my customers, it will be very nice when these issues are fixed.

miguelgrinberg, 2022-06-04:
This should work fine now, since files are streamed in chunks.

closed 2022-06-04 by miguelgrinberg

QGB, 2022-10-17:
```
Traceback (most recent call last):
  File "microdot_asyncio.py", line 367, in dispatch_request
  File "microdot_asyncio.py", line 421, in _invoke_handler
  File "mdot.py", line 113, in static
  File "microdot.py", line 631, in send_file
OSError: 20
Traceback (most recent call last):
  File "microdot_asyncio.py", line 367, in dispatch_request
  File "microdot_asyncio.py", line 421, in _invoke_handler
  File "mdot.py", line 113, in static
  File "microdot.py", line 631, in send_file
OSError: 20
```

miguelgrinberg, 2022-10-17:
@QGB the path and/or filename you are passing to send_file is likely wrong. Errno 20 is a "not a directory" error, and it happens in the line where the file is opened. Nothing to do with the actual sending of the file.