miguelgrinberg/microdot #14: Check for http headers must not be case-sensitive

zagl, 2020-10-21:
In the __init__ and create functions of Request the header keys are compared with case-sensitive strings (Content-Length, Content-Type and Cookie) but the HTTP specification says that these header keys are case-insensitive.

https://github.com/miguelgrinberg/microdot/blob/c5e1873523b609680ff67d7abfada72568272250/microdot/microdot.py#L90

The check should be case-insensitive like so:
`if header.lower() == 'content-type':`

I came across this issue because a HTTP client was using the HTTP header 'Content-type'.

miguelgrinberg, 2020-10-21:
Yes, it would be better to not assume the standard spelling.

steve1aa, 2022-01-25:
Same issue, is it possible to have a hot fix?

miguelgrinberg, 2022-01-25:
I you feel like contributing this fix I'll be happy to review it and merge it. If not, I am planning to work on several outstanding issues (this one included) in the near future, so there will be an update within the next month or two.

dpgeorge, 2022-02-16:
Looks like this was fixed in e16fb94b2d1e88ef681d70f7f456c37ee9859df6

miguelgrinberg, 2022-02-16:
Yes, thanks.

closed 2022-02-16 by miguelgrinberg