miguelgrinberg/microdot #15: Variable Rules in Microdot?

colin-infersens, 2020-12-09:
Hi Miguel,

I'm trying to set up variable rules for app routes, and I'm not succeeding in my efforts to get them to work. Would you expect the following to work, as an example, for any files stored in a sub directory ('/img/')?:

`@app.route("/img/<path:subpath>", methods=["GET", "POST"])
def content(subpath):
    filepath = 'www/img' + str(subpath)
    response = send_file(filepath)
    return response
`
Many thanks,

Colin

miguelgrinberg, 2020-12-10:
Yes, the route spec seems correct to me. But I think you are missing a slash when you build your file path:

    filepath = 'www/img/' + str(subpath)

A beginning slash is not part of your `subpath` pattern, so you have to add that.

colin-infersens, 2020-12-10:
Thanks for responding so quickly, Miguel; I'll give that a try.

Kind regards,

Colin

colin-infersens, 2020-12-10:
Hi Miguel,

I've tried adding the slash, as you suggested, as follows:

```
@app.route("/img/<path:subpath>", methods=["GET", "POST"])
def content(subpath):
    filepath = 'www/img/' + str(subpath)
    response = send_file(filepath)
    return response
```
...but I get the following error in the console:
```
Traceback (most recent call last):
  File "/lib/microdot_asyncio.py", line 115, in dispatch_request
  File "/lib/microdot_asyncio.py", line 153, in _invoke_handler
TypeError: function got multiple values for argument 'subpath'
GET /img/logo.jpg 500
```

Thanks again,

Colin

miguelgrinberg, 2020-12-10:
@colin-infersens Okay, the problem is that you did not include the required `request` argument in your handler. Microdot is fairly similar to Flask, but it does not have any of that app/request context magic. The request is always the first argument sent to handlers.

```
@app.route("/img/<path:subpath>", methods=["GET", "POST"])
def content(request, subpath):
    filepath = 'www/img/' + str(subpath)
    response = send_file(filepath)
    return response
```

colin-infersens, 2020-12-10:
Thank you, Miguel - that worked perfectly!

closed 2020-12-10 by colin-infersens