miguelgrinberg/microdot #16: Unhandled EINTR exception thrown from Microdot.run() accept() call

amakaro, 2021-02-04:
Microdot 0.3.0 (running on MicroPython v1.12-1 on 2020-02-09; linux version on Ubuntu 20.04.2 LTS) does not handle OS Error 4 (EINTR) that can be thrown from the s.accept() call in Microdot.run().
[Symptom: Hit Reload repeatedly in web-browser to load one HTML file and one PNG image file; EINTR gets thrown (seemingly at random) approximately 1 time in 10]. 
Arguably this exception should be handled by Micropython. 

This page ( https://stackoverflow.com/questions/14136195/what-is-the-proper-way-to-handle-in-python-ioerror-errno-4-interrupted-syst ) suggests the EINTR can be ignored and the operation retried.

I am working around this issue by a change to Microdot.run() as follows:

    def run(self, host='0.0.0.0', port=5000, debug=False):
        self.debug = debug

        s = socket.socket()
        ai = socket.getaddrinfo(host, port)
        addr = ai[0][-1]

        if self.debug:  # pragma: no cover
            print('Starting {mode} server on {host}:{port}...'.format(
                mode=concurrency_mode, host=host, port=port))
        s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
        s.bind(addr)
        s.listen(5)

        while True:
### Original code commented out
#           sock, addr = s.accept()
#           create_thread(self.dispatch_request, sock, addr)
### Replacement code
            try:
                sock, addr = s.accept() # Can throw OSError 4 (EINTR)
            except OSError as e:
                if e.args[0] != 4:  # Kludge, EINTR is not defined in errno
                    raise
            else:
                create_thread(self.dispatch_request, sock, addr)  
### End replacement code

miguelgrinberg, 2021-02-04:
Thanks! Looks like a good fix, I'll test it and add it to the package.

miguelgrinberg, 2021-02-04:
@amakaro What version of MicroPython are you using? It appears this was fixed directly in the interpreter in the 1.14 release. The commit is here: https://github.com/micropython/micropython/commit/9418611c8aa860812e28b6e2b6bfde7be5817b43.

Given that this is now handled internally I don't see a need to patch it again here.

amakaro, 2021-02-05:
Hi Miguel,

As per the original issue report, I was using Micropython 1.12. (“sudo apt install micropython” on Ubuntu 20.04).

 

I’ll have to see if I can get a 1.14 version installed on Ubuntu for dev purposes.

My target machine is an ESP32 and I see that a Micropython 1.14 build was released for the ESP32 only 3 days ago….

 

I agree that you do not need to patch; perhaps just document the dependency.

 

Adrian

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2021-04-06 by miguelgrinberg