miguelgrinberg/microdot #17: Example for using microdot_asyncio

kloknibor, 2021-03-23:
Hi,

thanks for the library! I am looking for a nonblocking webserver and found your project. I am pretty new to micropython though.

Would it be possible to get a small example of how to use microdot_asyncio so I can create a nonblocking web server?

Thanks in advance!

miguelgrinberg, 2021-03-23:
There isn't really big difference, the examples in this repository can be adapted simply by importing from `microdot_asyncio` instead of `microdot` and awaiting any methods that are coroutines. Here is the "hello" example ported to asyncio:

```python
from microdot_asyncio import Microdot, Response

app = Microdot()

htmldoc = """<!DOCTYPE html>
<html>
    <head>
        <title>Microdot Example Page</title>
    </head>
    <body>
        <div>
            <h1>Microdot Example Page</h1>
            <p>Hello from Microdot!</p>
        </div>
    </body>
</html>
"""

@app.route("", methods=["GET", "POST"])
def serial_number(request):
    print(request.headers)
    return Response(body=htmldoc, headers={"Content-Type": "text/html"})

app.run(debug=True)
```

Your handlers can be async functions (if you need to await on any coroutines) or standard functions if you don't.

kloknibor, 2021-03-24:
Probably a stupid question but how do I do the following:

Keep listening for a connection on the specified port. When there is no connection just keep running some_other_code(), when there is a connection server the webpage from htmldoc and then as soon as it is served continue with some_other_code() ?

kloknibor, 2021-03-24:
Basically like this

```
import select
from .UartHelper import UartConnection
from .WebComponent import WebComponent

class Main:
    def __init__(self):
        self.uart_status = {}

    def loop(self):
        uart_interface = UartConnection()
        web = WebComponent()

        while True:
            # get UART data
            uart_interface.check_uart()
            self.uart_status = uart_interface.return_status()
            print(self.uart_status)

            Microdot_server_keep_listening_for_connection()
           when_connection_serve_page()

  

```

miguelgrinberg, 2021-03-24:
The Microdot server takes control of your application when you call `app.run()`, this is in the style of the Flask framework for standard Python. If you want to do other stuff in parallel with this you can start async functions before calling `app.run()` and release control to the web server.

A better option would be to copy the code from the `run()` method and implement the server loop in a way that is more convenient to your needs.

miguelgrinberg, 2021-06-06:
I have added an asyncio example now.

closed 2021-06-06 by miguelgrinberg