miguelgrinberg/microdot #18: Possible to redirect from a alias (app name) to the host as a IP?

dubaleeiro, 2021-06-03:
Hi,

Thanks a lot for the library, first of all. This is actually a question and not a issue _per se_, (sorry for that). 

Instead of browsing for the ip adress of the host (like 192.168.x.y:5000), can I give a app name and get it redirected to the webserver (like http://appname.local)?

I looked up in some forums which are telling to include the "appname" to to "etc/hosts" file, which off course does not exist in a ESP8266...

Thanks!

miguelgrinberg, 2021-06-03:
The name goes in the hosts file of the client, not the server. So this should be done in the computer in which you run the browser that connects to the device.

dubaleeiro, 2021-06-04:
Thanks a lot for your fast response, Miguel.

I managed to change the host name using `sta.config(dhcp_hostname='Hostname')` in a ESP32. For wathever reason I don't understand it does not work with ESP8266 :(

Now I don't need to use the IP adress anymore, just `Hostname.local:5000` !

closed 2021-06-04 by dubaleeiro