miguelgrinberg/microdot #19: Feature Request: Shutdown server

jonathanfoster, 2021-06-03:
I have a use case that requires I start and stop the app based on certain criteria. For instance, if wifi credentials are invalid, I start the app and provide an API for updating the credentials. Once the credentials are updated, I stop the app and connect to the access point. I also allow the user to enter setup mode manually, which then restarts the app. Currently, the only way to accomplish this is to restart the entire process.

It would also be nice to leverage the existing event loop instead of running forever. Creating a task on the existing event loop is more inline with other MicroPython libs and is much more flexible in general.

miguelgrinberg, 2021-06-03:
Stoping the server is possible, but what do you mean by this:

> It would also be nice to leverage the existing event loop instead of running forever. Creating a task on the existing event loop is more inline with other MicroPython libs and is much more flexible in general.

I don't really understand what you mean here. This is a server, it is supposed to run forever, or at least expected to stay up for a long time. What do you mean by "existing event loop" here? Is this a reference to the asyncio event loop? Because that is already supported.

jonathanfoster, 2021-06-03:
In my case, I have another task running that's monitoring wifi settings and once they're set, it attempts to connect. So I need to run the app as well as the monitor task. I actually implemented this in my project. Shouldn't take too long to add the code here. Give me a few minutes and I'll submit a PR and you can check it out.

jonathanfoster, 2021-06-04:
@miguelgrinberg Here's what I was referring to by [not running forever](https://github.com/miguelgrinberg/microdot/pull/20/files#diff-edbcc046172ca7c3f103cf6d5163b197eda59e35cd062b69299f5aded3e98e6eR111). `asyncio.run_forever` is blocking so nothing after the `run` call will execute. By using `asyncio.create_task` instead, then you can call `run` whenever you want (assuming `asyncio.run` is called at some point).

This also allows the addition of a graceful shutdown. Since we now have access to the start server task, we can [await it and access the underlying server object](https://github.com/miguelgrinberg/microdot/pull/20/files#diff-edbcc046172ca7c3f103cf6d5163b197eda59e35cd062b69299f5aded3e98e6eR120). From there we just close the underlying _serve task, which closes the socket and releases the address.

Hopefully this makes sense. I figured it would be easier to explain with working code.

miguelgrinberg, 2021-06-04:
Okay, seeing the code in your PR helps me understand what you mean better.

I think you are confusing two different things. I understood that you want the ability to stop the server. What wasn't clear to me before is that you'd also like to run the server on an already running loop, instead of in a loop owned by the server.

Your PR mixes these two separate features, which is a problem, first because I can't add a feature in the async server and not have it in the sync server, and second, because there is no reason why the server can't also be stopped when using the `run_forever` method.

Thanks, now I understand what you are asking for, let me see if I can come up with something that works for you while also preserve the parity between my two server implementations.

jonathanfoster, 2021-06-04:
Gotcha. Yeah, a better way to describe it would be to run the server without blocking.

The key thing with stopping the asyncio server is that you need to cancel the `_serve` task to actually close the socket and release the port. 

I haven’t used the threaded server yet. I’ll take a look as well.

By the way, thanks for creating Microdot. It’s hands down the best HTTP server for MicroPython.

closed 2021-06-04 by miguelgrinberg

miguelgrinberg, 2021-06-04:
@jonathanfoster I've added an async example now, including a shutdown option: https://github.com/miguelgrinberg/microdot/blob/master/examples/hello_async.py.

Let me know if this works.

jej, 2022-01-22:
After few months away, I am happy to discover the `start_server()` method. I had the same need as @jonathanfoster to run the webserver in parallel to another existing loop.

I would just comment about the name `start_server`, that is not semantically correct to me. We don't have `run_server` or `shutdown_server` methods because class context is the server already.

I use it this way:

```
webserver = Microdot()

@webserver.route('/')
...

asyncio.create_task(webserver.start_server())

try:
    asyncio.run(myloop())
...
```
If `Microdot()` is really the webserver, the method should be named simply `start` to have `webserver.start()`, like other methods.

It's a detail but I was a bit disturbed by the name, and went to the code to get it clear. I guess it's easily possible to alias this method (`self.start = self.start_server`) to manage a transition if needed.

Concerning the `shutdown` I am not sure it works everytime. I have some `OSError: [Errno 112] EADDRINUSE` very often. Is there any `uasyncio.Server.close() / uasyncio.Server.wait_closed()` missing?

Hope it helps. Thanks @miguelgrinberg for the great work.

jej, 2022-01-22:
PS:
The way it's done in tinyweb (https://github.com/belyalov/tinyweb/blob/33d73c90f005cfd3423affe358cad0f13c37728f/tinyweb/server.py#L674), with a `loop_forever=False` parameter:

```
    def run(self, host="127.0.0.1", port=8081, loop_forever=True):
        """Run Web Server. By default it runs forever.
        Keyword arguments:
            host - host to listen on. By default - localhost (127.0.0.1)
            port - port to listen on. By default - 8081
            loop_forever - run loo.loop_forever(), otherwise caller must run it by itself.
        """
        self._server_coro = self._tcp_server(host, port, self.backlog)
        self.loop.create_task(self._server_coro)
        if loop_forever:
            self.loop.run_forever()
```

miguelgrinberg, 2022-01-22:
@jej I have updated the hello_async.py example to use `app.run()` to start the server, which is more concise. Not sure about the shutdown failure, if you have a method I can use to reproduce it I can debug it here.

jej, 2022-01-22:
I am not sure we speak about the same thing. My comment was about the way it's implemented, not the use case.

miguelgrinberg, 2022-01-22:
I'm sorry but I don't understand. You said you didn't like the name `start_server`, you didn't say anything about implementation. The `run` method is actually a simpler method to start that maybe you prefer. I'm not going to change `start_server` as that is used in asyncio servers beyond this package.

jej, 2022-01-22:
Yep, I just spotted that `server.start_server()` on a server object is not really useful. Maybe `server.start()` is ok... like you did for `server.run()`, `server.shutdown()` etc...

And the tinyweb implementation avoid the separate method for the same think, with an additional arg.

miguelgrinberg, 2022-01-22:
@jej it's just a name, and many other asyncio servers use it, so I'm not going to change it. Sorry.

jej, 2022-01-22:
> I'm not going to change `start_server` as that is used in asyncio servers beyond this package.

Ok, no problem. I thought it was independant of anything else, just a choice to solve the first request above.

Thanks.

miguelgrinberg, 2022-01-23:
There's a lot of stuff in this thread, I'm sorry but I don't know what "request above" you refer to. If you have an issue with this package beyond naming things, please open a separate issue.

jej, 2022-01-23:
If I understood jonathanfoster's request (ie. the first request above): "It would also be nice to leverage the existing event loop instead of running forever."
But we can stop here, no problem.
Have a good continuation.

miguelgrinberg, 2022-01-23:
@jej `start_server` is a coroutine. You can await it, and that will start the server and return when the server is shutdown. Setting aside the name, isn't that exactly what the OP asked? This was done specifically per this request.

jej, 2022-01-23:
Sorry, this is a dialogue of the deaf. I prefer to delete my messages to avoid confusion for other users. Forget it, it's not a problem.

miguelgrinberg, 2022-01-23:
@jej sorry you feel that way, but as you say, it's not problem. In that we agree. Keep the following in mind:

- This project is open source. If there is anything you don't like, you are welcome to change it in your copy. If you like `app.start()` better than `app.start_server()`, by all means edit your copy and make it be the way you like.
- If you like the way picoweb starts the server, then either use picoweb (if it works in your project) or else as above, adapt their `app.run(loop_forever)` solution to your copy of Microdot. The fact that you like it better doesn't mean that I should like it better too, Microdot is not a copy of picoweb, and is not based on it. My inspiration for this project is Flask, not picoweb.
- If you still have problems with errors during shutdown, then file a separate issue and describe the problem in detail so that I can reproduce it.

jej, 2022-01-23:
Don't worry, no problem. It was just a small remark at the beginning, but the misunderstanding generated unnecessary noise. Sorry for that, maybe I was no clear enough. I purge the thread because no reason to pollute this PR, and don't want to cause more off topic answers. Cheers.

commit 8b82ffae (2021-06-04): Server shutdown (Fixes #19)
commit 0ad538df (2021-06-04): Server shutdown (Fixes #19)