miguelgrinberg/microdot #21: HTML multiple dropdown list returns only the last selected value

dubaleeiro, 2021-06-04:
Hi again,
I've noticed that HTML select/options list with multiple selection returns only the last selected value in the `request.form`:

`<form method="POST" action="">
<p>
<i class="bi bi-calendar-day"></i>
<label for="weekdays">days:</label>&nbsp;&nbsp;
<select class="form-select" name="weekdays" id="weekdays" multiple>
<option value="" selected hidden disabled>change</option>
<option value="0">Monday</option>
<option value="1">Tuesday</option>
<option value="2">Wednesday</option>
<option value="3">Thursday</option>
<option value="4">Friday</option>
<option value="5">Saturday</option>
<option value="6">Sunday</option>
</select></p>
<button type="submit" name="settings" value="save" class="btn btn-primary">
</form>`

Is this an expected/known behavior? Any hints on how to fix it? Thanks!

miguelgrinberg, 2021-06-04:
I never got around to implement duplicate query or form values, as I haven't needed them myself. I'll look into it.

closed 2021-06-05 by miguelgrinberg

dubaleeiro, 2021-06-05:
Great, thanks !
Is it possible to use it with microdot_asyncio.py as well?

miguelgrinberg, 2021-06-05:
Yes, both have this now. Note that you will need to use the `getlist()` method to get your options:

    selections = request.form.getlist('weekdays')

dubaleeiro, 2021-06-06:
Thanks !
Any hints on how to solve this error, please?
`AttributeError: 'dict' object has no attribute 'getlist'`

miguelgrinberg, 2021-06-06:
@dubaleeiro did you upgrade to the latest version? I'm just now about to publish it to PyPI, the change was only here in the main branch until now.

dubaleeiro, 2021-06-06:
Yes, I have copied both microdot.py and microdot_asyncio.py files to my ESP32 ports and made a new build. I will double check it if it really worked then.

miguelgrinberg, 2021-06-06:
@dubaleeiro check if there is a `MultiDict` class in your micropython.py file. If you have that, then I guess I missed some use case.

dubaleeiro, 2021-06-06:
There was a problem with my build, now I see the `MultiDict` class in the `micropython.py` file. Unfortunately I came across a different problem when testing the `hello_async.py`:

`AttributeError: 'Server' object has no attribute 'task'`

Details:
`Starting async server on 0.0.0.0:5000...
Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 37, in main
  File "microdot_asyncio.py", line 165, in start_server
  File "uasyncio/stream.py", line 1, in wait_closed
AttributeError: 'Server' object has no attribute 'task'`

miguelgrinberg, 2021-06-06:
This looks like a problem with your uasyncio library. You may want to make sure that you have all the files for this package, and that they all come from the same (latest) version.

dubaleeiro, 2021-06-06:
I am very intrigued by this problem... I have cloned the micropython repository (with asyncio off course) and the error persists. 

I have also noticed you haven't changed the `microdot_asyncio.py`, right? The normal webserver based on `microdot.py` works fine.

dubaleeiro, 2021-06-06:
Unfortunately I could not trace-back the problem by looking at the asyncio standard libraries. That has obviously to do with the fact I am just a beginner in Python...
I have therefore opened the following [issue](https://github.com/peterhinch/micropython-async/issues/63).

miguelgrinberg, 2021-06-06:
I'll take another look later today. Maybe I did forget to change something in the async version.

miguelgrinberg, 2021-06-06:
@dubaleeiro Give the code another try. I've added a workaround for this bug.

dubaleeiro, 2021-06-06:
Sorry, unfortunately the error persists:

`Starting async server on 0.0.0.0:5000...
Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 37, in main
  File "microdot_asyncio.py", line 165, in start_server
  File "uasyncio/stream.py", line 1, in wait_closed
AttributeError: 'Server' object has no attribute 'task'`

miguelgrinberg, 2021-06-06:
@dubaleeiro you haven't updated microdot_asyncio.py to the latest from this repository. Line 165 (as shown in your error) corresponds to the old version. My change moved the lines, in the current version line 165 is a while loop, not a line that could cause this error now.

dubaleeiro, 2021-06-06:
Indeed - I got confused with my custom builds, I am sorry.

Now it works like a charm and I can get all multiple elements from the drop-down list. Thank you very much !!

commit b0c25a1a (2021-06-05): Support duplicate arguments in query string and form submissions

Fixes #21