miguelgrinberg/microdot #22: Error installing 'microdot-asyncio': , packages may be partially installed

jonathanfoster, 2021-06-05:
I tried to install microdot-asyncio 0.4.0 and got the following error:
```shell
$ micropython -m upip install -p lib microdot-asyncio
Installing to: lib/
Warning: micropython.org SSL certificate is not validated
Error installing 'microdot-asyncio': , packages may be partially installed
```

Looks similar to https://github.com/miguelgrinberg/microdot/issues/4.

jonathanfoster, 2021-06-05:
Yup, looks like the wheel file issue again. The other versions only include a source file.

```shell
$ curl -s https://pypi.org/pypi/microdot-asyncio/json | jq '.releases["0.4.0"]'
[
  {
    "comment_text": "",
    "digests": {
      "md5": "31359197c34a7c11e4aa8bcfa89d9775",
      "sha256": "bf8b93cfb25c738ca6e124238c066184b1a252626473ed340633e793a397c20a"
    },
    "downloads": -1,
    "filename": "microdot_asyncio-0.4.0-py2.py3-none-any.whl",
    "has_sig": false,
    "md5_digest": "31359197c34a7c11e4aa8bcfa89d9775",
    "packagetype": "bdist_wheel",
    "python_version": "py2.py3",
    "requires_python": null,
    "size": 3197,
    "upload_time": "2021-06-04T16:16:26",
    "upload_time_iso_8601": "2021-06-04T16:16:26.674393Z",
    "url": "https://files.pythonhosted.org/packages/52/06/f44d90278ff7cfadcc13b6ecbdaea4fd363be171a7e2a1554b4a5cffe7a5/microdot_asyncio-0.4.0-py2.py3-none-any.whl",
    "yanked": false,
    "yanked_reason": null
  },
  {
    "comment_text": "",
    "digests": {
      "md5": "ccaa39b63111e9d07479337a67f2d9e5",
      "sha256": "914adb88ef6bdf075a76185d684f9aed39c07672c276aa8b64122f5a41b4eee7"
    },
    "downloads": -1,
    "filename": "microdot-asyncio-0.4.0.tar.gz",
    "has_sig": false,
    "md5_digest": "ccaa39b63111e9d07479337a67f2d9e5",
    "packagetype": "sdist",
    "python_version": "source",
    "requires_python": null,
    "size": 2966,
    "upload_time": "2021-06-04T16:16:27",
    "upload_time_iso_8601": "2021-06-04T16:16:27.632515Z",
    "url": "https://files.pythonhosted.org/packages/ba/a7/fadbd9c81769beab2dd57edcb5ba44c69d60ec8bc843011fad653a3f50b4/microdot-asyncio-0.4.0.tar.gz",
    "yanked": false,
    "yanked_reason": null
  }
]
```

miguelgrinberg, 2021-06-05:
Oh yeah, I always forget. I use the same script to publish to PyPi for all my packages, this one is special in that it does not support the wheel. Sorry about that, I'll delete them in a second.

miguelgrinberg, 2021-06-05:
Okay, should be fine now. I need to find a definitive solution to this problem, because I don't like to have to suppress the wheel packages forever.

closed 2021-06-05 by miguelgrinberg