miguelgrinberg/microdot #25: Making the Response reason available to the application

dxf19, 2021-08-11:
I recommend a simple enhancement to the Response class to allow for specification of the _reason_.  Currently it hard-codes this to "OK" for a 200 response and N/A for everything else.  For example, if returning a 400 response, it would be more useful for the application to specify _why_, where there is provision for it, instead of having to put this in the response body.  It is apparently quite valid to customise this, per https://www.ibm.com/docs/en/cics-ts/5.1?topic=concepts-status-codes-reason-phrases - "The HTTP/1.1 specification says that the reason phrases for each status code may be replaced by local equivalents."

A suggested fix is to replace, in microdot.py...
```
164:     def __init__(self, body='', status_code=200, headers=None):
165:         self.status_code = status_code
```
with
```
164:     def __init__(self, body='', status_code=200, reason="N/A", headers=None):
165:         self.status_code = status_code
166:         self.reason = reason
```
and
```
210:             reason='OK' if self.status_code == 200 else 'N/A').encode())
```
with
```
211:             reason='OK' if self.status_code == 200 else self.reason).encode())
```

miguelgrinberg, 2021-08-11:
Thanks for the suggestion. I have implemented it, more or less as you suggested.

closed 2021-08-11 by miguelgrinberg

bulletmark, 2021-08-11:
Thanks both of you for this. I am developing a service which returns 400 code for various client side errors so I was returning the error text in the body. I was wondering why Chrome developer console always showed only a red N/A with the error and I had to dig down and expand out the response body to see the actual text. Now the error text formats nicely in red within the console!

commit bd74bcab (2021-08-11): Accept a custom reason phrase for the HTTP response (Fixes #25)