miguelgrinberg/microdot #26: Request fails when uploading large files

bulletmark, 2021-09-18:
I believe this issue is somewhat analogous to issue #3 and #5. I am trying to upload a large file from my browser [using a simple XMLHttpRequest()](http://pedrolobito.com/demos/html5_file_upload) to an ESP32 running async Microdot. However, the code dies immediately within `Request.create()` at the line `body = await client_stream.read(content_length)` because it is trying to immediately allocate memory for the full file size. I was expecting that for this case the request body would be a stream which I could read in block chunks. Am I missing something?

I guess ideally there would be a `request.files` attribute like you describe for flask [here](https://blog.miguelgrinberg.com/post/handling-file-uploads-with-flask). Any plans for that?

miguelgrinberg, 2021-09-18:
The current version of the request object does not support large files, because as you've seen, loads the response into memory. I'm not sure I want to create something as elaborate as the request.files in Flask, but I can look into exposing the stream to the endpoint so that it can consume it without an intermediate memory storage.

bulletmark, 2021-09-18:
Thanks. Note this is not an esoteric request as I want to implement wifi OTA firmware image update via my web interface which I guess would be a common desire among MicroPython Microdot users. So I don't actually need to be able to write to the file system. I want to write blocks directly to the [ESP32 Partition](https://docs.micropython.org/en/latest/library/esp32.html#esp32.Partition.writeblocks) so yes, having access to the stream handle would be perfect for that.

steve1aa, 2021-11-05:
I have the same situation. is there a workaround? I am using the master branch.

ser, 2022-01-01:
It's also important to me, as the files served from my audio microcontroller can be large.

dpgeorge, 2022-02-16:
I would also like this feature, to be able to stream in large files in a request.

Eg: if the content-length is larger than some threshold then set `request.body = client_stream`.

Also if the content-length is not given then there could be a body, in which case `request.body = client_stream` would also be useful.

commit 3464cd85 (2022-02-18): Support streamed request payloads (Fixes #26)
closed 2022-02-18 by miguelgrinberg