miguelgrinberg/microdot #28: Is there a way to do a "catchall" for captive portals?

UnexpectedMaker, 2021-09-24:
Hey @miguelgrinberg - I'm loving microdot, thanks heaps for making it.

Is there a way to do a catchall route? captive portals have so many different endpoints on different OS versions, and when I was using tinyweb, it had a catchall that enforced any url that didn't have a route specified to go to a specific route of my choosing.  

Thanks.

miguelgrinberg, 2021-09-24:
Yes! This is done like you do it in Flask, with the `path` URL pattern. Here is a quick example:

```python
@app.route('/<path:path>')
def hello(request, path):
    # ...
```

UnexpectedMaker, 2021-09-24:
Ahh, that works great, thanks!
I'm a Django guy... not flask, so I have never seen that format before :)

closed 2021-09-24 by UnexpectedMaker