miguelgrinberg/microdot #35: Arduino RP2040 nano connect - SOL_SOCKET

JohannesMaierhofer, 2022-02-16:
Running microdot on the official Micropython 1.18 firmware results in a error on line:
`self.server.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)`
Unfortunately, commenting out this line does not help.

miguelgrinberg, 2022-02-16:
Without the error I cannot really tell you what the problem is.

JohannesMaierhofer, 2022-02-16:
sorry, here is the full output. I could also imagine that it is more an issue on the port of Micropython to the RP2-Family.
```
Starting async server on 0.0.0.0:5000...
Traceback (most recent call last):
  File "<stdin>", line 48, in <module>
  File "microdot_asyncio.py", line 216, in run
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "microdot_asyncio.py", line 177, in start_server
  File "uasyncio/stream.py", line 1, in start_server
AttributeError: 'module' object has no attribute 'SOL_SOCKET'
```

miguelgrinberg, 2022-02-16:
The error is in the `start_server` function of the `uasyncio` module. Not sure what kind of networking the MicroPython builds for this device have, but it appears `uasyncio` isn't compatible with it at this time.

JohannesMaierhofer, 2022-02-16:
Thanks for your reply. I already tried with the threaded version, but had no luck there too. I could start the server, but the first request (using curl) kicked the following issue: 

```
Starting threaded server on 0.0.0.0:80...
Traceback (most recent call last):
  File "<stdin>", line 21, in <module>
  File "microdot.py", line 770, in run
OSError: -1
>Traceback (most recent call last):
  File "microdot.py", line 811, in dispatch_request
  File "microdot.py", line 257, in create
  File "microdot.py", line 318, in _safe_readline
OSError: [Errno 1] EPERM
Unhandled exception in thread started by <closure>
Traceback (most recent call last):
  File "microdot.py", line 36, in run
  File "microdot.py", line 860, in dispatch_request
  File "microdot.py", line 407, in write
OSError: [Errno 1] EPERM
```

The minimum example-code is:
```
import network,socket

SSID ='My_Nano_RP2040_Connect'   # Network SSID
KEY  ='1234567890'  # Network key (must be 10 chars)

wlan = network.WLAN(network.AP_IF)
wlan.active(True)
wlan.config(essid=SSID, key=KEY, security=wlan.WEP, channel=2)
print("AP mode started. SSID: {} IP: {}".format(SSID, wlan.ifconfig()[0]))

from microdot import Microdot

app = Microdot()

@app.route('/')
def hello(request):
    return 'Hello, world!'

app.run(port=80, debug=True)
```

I would love to use your package on the RP2040 connect as it is exactly what I was searching for.

miguelgrinberg, 2022-02-16:
Unfortunately I have nothing comparable to your device to test on.

JohannesMaierhofer, 2022-02-17:
Can I provide you anything useful? Trying something to give you a clue what could go wrong?

miguelgrinberg, 2022-02-17:
You should ask the MicroPython people, because the build that you are using is missing things. This is unlikely to be a problem with this project.

miguelgrinberg, 2022-08-07:
Microdot now defaults to running requests synchronously, so this shouldn't be a problem anymore. Please give the latest release of microdot.py a try and reopen if you still have problems.

closed 2022-08-07 by miguelgrinberg