miguelgrinberg/microdot #40: Feature request: Websocket support

beyonlo, 2022-04-30:
Hello! 

As you [wrote here](https://github.com/miguelgrinberg/microdot/issues/39)  that are opened to suggestions for new features, I have just one that I consider very important: the Websocket support.

My goal is to create a simple web interface for the ESP32, STM32  to configure some sensors (like as temperature, humity, etc) and monitoring (events, alarms) it real time. Well, with Websocket I can just wait a event/alarms coming without  todo a polling on the HTTP Server.

In my opinion is a very important feature!

Thank you in advance!

miguelgrinberg, 2022-04-30:
Duplicate of #34, also by yourself. Closing.

closed 2022-04-30 by miguelgrinberg

beyonlo, 2022-05-01:
Hello @miguelgrinberg 

Ohh, I'm sorry, I forgot that I already did that. I was just looking for the opened issues, and that request was at he closed issues.

Please, if you in some moment think to start a Websocket support for the Microdot, let us to know :)

Thank you very much!