miguelgrinberg/microdot #41: Local use of CSS on the gpio.html example

beyonlo, 2022-05-06:
Hello!

Maybe this is a very basic question, but I have no idea how is possible to do that.

I was running the **gpio.html** example. It is very good, but I would like that all CSS `bootstrap.min.css` stay inside microcontroller flash, how is possible?

I downloaded the `bootstrap.min.css` and copied into microcontroller flash, in the same directory where there are the `gpio.py` and `gpio.html`, and changed this line of `gpio.html`:

`        <link rel="stylesheet" href="https://stackpath.bootstrapcdn.com/bootstrap/4.3.1/css/bootstrap.min.css" integrity="sha384-ggOyR0iXCbMQv3Xipma34MD+dH/1fQ784/j6cY/iJTQUOhcWr7x9JvoRxT2MZw1T" crossorigin="anonymous">
`

and substituted by this:

`<link rel="stylesheet" type="text/css" href="bootstrap.min.css">.
`

But that do not works.

I tried to change the path on the `href` too, with these options, but nothing works:
```
<link rel="stylesheet" type="text/css" href="/bootstrap.min.css">.
<link rel="stylesheet" type="text/css" href="./bootstrap.min.css">.

``` 
Could you please tell me what I need to do to that works?

I readed the docs https://microdot.readthedocs.io/en/latest/api.html# to find something, but no success. I would also like to inform you that I found a simple error in that documentation. In the https://microdot.readthedocs.io/en/latest/api.html#microdot.Microdot.after_request the example was wrote as `@app.before_request`

Thank you very much.

miguelgrinberg, 2022-05-06:
Microdot does not have default support for serving static files, you have to implement this yourself. If you only have one file that you want to serve, then you can add a custom route for it:

```python
@app.route('/bootstrap')
def bootstrap(request):
    return send_file('bootstrap.min.css')
```

And then in your HTML you can use something like this:

```html
<link rel="stylesheet" type="text/css" href="/bootstrap">
```

beyonlo, 2022-05-10:
Hello @miguelgrinberg 

> ```python
> @app.route('/bootstrap')
> def bootstrap(request):
>     return send_file('bootstrap.min.css')
> ```
> And then in your HTML you can use something like this:
> ```
> <link rel="stylesheet" type="text/css" href="/bootstrap">
> ```

That works fine, thanks! But there is a problem: everytime that I click on Button `Read`, `Set Low` and `Set high`, the CSS is loaded again - that turn usability a bit bad/slow. Is there a way to load just once the CSS?

Thank you in advance!

miguelgrinberg, 2022-05-10:
You can add a `Cache-Control` header to the response so provide hints to the client regarding how long to cache the file for. But ultimately it is up to the client to decide when to request the file again and when to take it from the cache.

miguelgrinberg, 2022-08-07:
A `static.py` example has been added to show how a Microdot application can serve static files from a directory.

closed 2022-08-07 by miguelgrinberg