miguelgrinberg/microdot #43: Error while import microdot

yura50525, 2022-05-17:
I am using node mcu esp8266.
When I import microdot lib there is error:

`ImportError: can't import name Microdot`

beyonlo, 2022-05-17:
@yura50525 

That looks like that the `microdot.py` and `microdot_asyncio.py` does not in the same directory that your `app.py` that you are testing. Please look at the https://github.com/miguelgrinberg/microdot/tree/main/src about those files.

And if you can, please, paste your entire terminal commands for better help, not just the output.

miguelgrinberg, 2022-05-17:
Try the import in the REPL. Does it work there?

I'm not sure if this is the problem or not, but I think it is odd to use `boot.py` for your code. Shouldn't you use `main.py` instead and leave `boot.py` alone?

yura50525, 2022-05-17:
@miguelgrinberg 

I already try to import in REPL but there is same error. 

I also tried to use main.py instead of boot.py there is an error to.

I think the  library file I download is broken or i don't upload it to esp8266 right.

yura50525, 2022-05-17:
Maybe you have another source from where library can be downloaded?

miguelgrinberg, 2022-05-17:
The files in this repo are not broken. If these did not work there would be lots of reports of problems, and yours is the only one.

This is actually easy to troubleshoot on your side. Open microblog.py and see if there is a `Microblog` class in it. That is what you are importing. If micropython says this symbol does not exist in the file, then you shouldn't be able to find it, right?

yura50525, 2022-05-17:
Right, thanks for your help😄

If I find a solution to my problem I will leave new comment.

Nixellion, 2022-05-27:
I'll have to +1 to this problem. I get the same error on a Wemos D1 Mini.

miguelgrinberg, 2022-05-27:
@Nixellion What is the file structure in your device? Please include all folders and files.

Nixellion, 2022-05-27:
@miguelgrinberg Tried a clean project with just `boot.py, main.py, microdot.py` in it. Removed everything from the board, uploaded.

```
>>> import os
>>> os.listdir()
['boot.py', 'main.py', 'microdot.py']
>>> import microdot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: 
>>> 
```

boot.py
```

# import webrepl
# webrepl.start()

from machine import Pin, reset

led = Pin(2, Pin.OUT)
reset_pin = Pin(0, Pin.OUT)
reset_pin.value(1)

try:
  import usocket as socket
except:
  import socket

import esp
esp.osdebug(None)

import gc
gc.collect()

# print("Reset pin value at boot:", reset_pin.value())
print("boot.py complete.")

```

`main.py` is empty in this case.

miguelgrinberg, 2022-05-27:
@Nixellion It's an out of memory error. You may need to compile or freeze the file and maybe it'll fit.

Nixellion, 2022-05-27:
@miguelgrinberg I tried precompiling pyc files, and placing them on the board. Without the `.py` version of microdot it gives this:

```
>>> os.listdir()
['boot.py', 'main.py', 'microdot_asyncio.py', 'microdot.pyc', 'precompile.py', '__pycache__']
>>> import microdot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'microdot'
```

But I'm not sure if this is the correct way of compiling it for an esp? What did you mean when you said to compile or freeze it?

miguelgrinberg, 2022-05-27:
You ended up with a `.pyc` file? Where did that come from? Should have been an `.mpy` file. See https://docs.micropython.org/en/latest/reference/mpyfiles.html.

Nixellion, 2022-05-27:
Tried to precompile with mpy-cross, but I guess I need to do more research into how to properly use it, got:

```
>>> import os
>>> os.listdir()
['boot.mpy', 'microdot.mpy', 'main.mpy']
>>> import microdot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: incompatible .mpy file
>>> 
```

Making a custom firmware build sounds like a hassle, but I'll look into it as well if I'll have time. 

I guess I'll use my own microserver for this project for now.

miguelgrinberg, 2022-05-27:
Maybe you used an mpy-cross that does not match the MicroPython firmware that you have on your device.

Another thing I just thought of is that you can also use an older release of microdot. Something like the 0.4.0 release, which has less stuff so is smaller.

Nixellion, 2022-05-28:
I'll try that when\if I have time.

Wemos d1 mini (and I think most other esp8266 boards) have 33K RAM available. Microdot.py is 32K. Maybe there's some relation there :)

closed 2022-08-07 by miguelgrinberg

brave-ulysses, 2022-12-10:
had similar issue - cleaning sys.path and removing multiple microdot files cleared it up

Sniphs98, 2023-07-06:
> I'll try that when\if I have time.
> 
> Wemos d1 mini (and I think most other esp8266 boards) have 33K RAM available. Microdot.py is 32K. Maybe there's some relation there :)

I also have a D1 and I'm experiencing the same issue. Has there been any solution found for this problem yet? I didn't quite understand the lower part as I'm still new to the whole microcontroller game.

If I come across a solution on another page, I will post it here as well.