miguelgrinberg/microdot #44: Support streaming contents

zumoshi, 2022-06-03:
Hello

I wanted to use [SSE](https://developer.mozilla.org/en-US/docs/Web/API/Server-sent_events), However, that requires leaving the connection open and being able to push/flush multiple responses for the same request.

The way Flask has implemented this feature is to allow a generator as the response body. (see: [Streaming Content](https://flask.palletsprojects.com/en/2.1.x/patterns/streaming/)). Which can be used to [implement SSE](https://maxhalford.github.io/blog/flask-sse-no-deps/).

I can see that you already have a similar setup in [body_iter](https://github.com/miguelgrinberg/microdot/blob/5b5eb907d83d94dde544b266e6659071e4d47ee1/src/microdot.py#L451), is there any chance you consider exposing that (i.e. allow the application to overwrite body_iter to any generator if a function is returned as response), so the application wouldn't be limited to only streaming file-like objects?

Thanks.

miguelgrinberg, 2022-06-03:
I'll look into it. You are correct, with the recent changes I've made this should be possible.

closed 2022-06-04 by miguelgrinberg

zumoshi, 2022-06-05:
Thanks for the speedy update. It works great.

Also learned something new from you, didn't know `multipart/x-mixed-replace` was a thing.

The only issue I have is getting traces on console everytime a connection is aborted, which happens more often with long running connections such as SSE.

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "c:\program files\python39\lib\threading.py", line 954, in _bootstrap_inner
    self.run()
  File "c:\program files\python39\lib\threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\bor\.virtualenvs\boz-ZIWTKwHQ\lib\site-packages\microdot.py", line 865, in handle_request
    res.write(stream)
  File "C:\Users\bor\.virtualenvs\boz-ZIWTKwHQ\lib\site-packages\microdot.py", line 458, in write
    stream.flush()
  File "c:\program files\python39\lib\socket.py", line 722, in write
    return self._sock.send(b)
ConnectionAbortedError: [WinError 10053] An established connection was aborted by the software in your host machine
```

I can see in the diff that you already have some logic to suppress `errno.ECONNABORTED`. but it seems that only works for Linux, and Windows requires a different way to handle/catch it.

commit 874fdea7 (2022-06-04): Stream responses (Fixes #44)
commit d71665fd (2022-06-04): Stream responses (Fixes #44)