miguelgrinberg/microdot #45: Question: How to handle CORS in Microdot?

steveli777, 2022-06-06:
Same to the title, I got the 404 error while the Microdot handled the OPTIONS request. 
Is there a way to enable CORS in Microdot or bypass it?

miguelgrinberg, 2022-06-06:
@steveli777 I would imagine the error that you get is a 405, not a 404?

Microdot does not handle the OPTIONS method, but it probably should. At this time your only option is to handle it yourself by returning the `Allow` request for the supported methods, and also any necessary CORS headers per your needs.

steveli777, 2022-06-06:
The error I got is 404. And it is a preflight request (OPTIONS). 

I found a workaround to handle the OPTIONS, not perfect, but it works. 

`
def not_found(req: Request):
    res = None
    if req.method == 'OPTIONS':
        res = Response(res)
        res.headers["Access-Control-Allow-Origin"] = '*'
        res.headers["Access-Control-Allow-Methods"] = '*'
        res.headers["Access-Control-Allow-Headers"] = '*'
        res.headers["Access-Control-Allow-Credentials"] = 'true'
        res.headers["Access-Control-Max-Age"] = '86400'
        return res
`

miguelgrinberg, 2022-06-06:
You need to add `OPTIONS` to the list of supported methods.

```python
@app.route('/foo', methods=['GET', 'OPTIONS'])
def foo():
    ...
```

This will prevent the 404, and if you have a before request handler, it will be invoked before your endpoint.

steveli777, 2022-06-06:
Aha, thanks for the suggestion.

closed 2022-08-07 by miguelgrinberg

benoitLecuppe, 2023-02-25:
could you add an @app.Options( request, [access-control-request-parameters,...]) ?

and some "use samples"

commit 67798f7d (2023-03-23): Cross-Origin Resource Sharing (CORS) support (Fixes #45)