miguelgrinberg/microdot #48: Stops working when idling a bit?

WaGi-Coding, 2022-07-15:
Hello, i've the simple hello world example running atm. using uasyncio on an ESP32-WROOM from AZ-Delivery.

My problem is that it works fine first, but when i let it idle a bit and try to open the page again, it is just stuck in loading the page and nothing happens anymore.

Used micropyserver first without this issue, so i might miss something related to this library or maybe "u"async is the problem?

Would really like to use this, but i've no clue how to solve that issue, i run the server in debug mode & it won't spit out any errors or so, just hangs somewhere. The page loading never ends like when unreachable or so. Network inspector says "Pending". Sometimes it happens after ~10 minutes, sometimes after 30 minutes :S

Any help would be appreciated

```py
import network
import socket
import time

import uasyncio
from microdot_asyncio import Microdot

wlan_sta = network.WLAN(network.STA_IF)

def do_connect():
    import network
    #sta_if = network.WLAN(network.STA_IF)
    if not wlan_sta.isconnected():
        print('connecting to network...')
        wlan_sta.active(True)
        wlan_sta.connect('myssid', 'mypass')
        while not wlan_sta.isconnected():
            pass
    time.sleep(1)
    print('network config:', wlan_sta.ifconfig())

app = Microdot()

do_connect()

status = wlan_sta.ifconfig()
addr = socket.getaddrinfo(status[0], 80)[0][-1]

@app.route('/')
async def index(req):
    return 'Hello, world!'

async def main():
    await app.start_server(host=addr[0], port=80, debug=True)

uasyncio.run(main())
```

miguelgrinberg, 2022-07-15:
I'm not sure what the problem might be. I guess I would make sure I have the latest uasyncio library, in case that is where the problem comes from.

Have you tried using the sync server option? That would be more in line with the other web server that you used and would provide another data point to determine if this is an async issue or something more general.

WaGi-Coding, 2022-07-15:
okay i currently have another problem, now i wanted to make changes but it hangs entirely, i cannot even erase flash anymore with esptool o.O
I'll try your suggestions on a new esp32, maybe this one was about to die.

APCBoston, 2022-08-06:
I'm having a similar issue.  First request gets handled correctly, but second and subsequent requests time out with no sign that the app is receiving them:

```
    response.default_content_type =  'text/html'
    
    app = Microdot()
    
    @app.route('/')
    def index(request):
        if debug: print("starting request")
        return 'Hello, World!'
    
    @app.route('/metrics')
    def metrics(request):
        return 'Metrics bit'
    
    @app.before_request
    def start_timer(request):
        request.g.start_time = time.time()

    @app.after_request
    def end_timer(request, response):
        duration = time.time() - request.g.start_time
        print(f'Request took {duration:0.2f} seconds')
    
    app.run()
```
Sending a GET request to the server via browser or curl, I get the correct "Hello, World!" response on the first go, as well as the expected output to stderr... but then if I immediately send a second request... nada.  Times out after 6700 ms on the client side; no output on the server side.  And I have to reset the board to get it going again -- it's not enough to interrupt and restart the program.

miguelgrinberg, 2022-08-07:
@APCBoston This is probably unrelated to the original issue, but I have tested your code on a Pico W and was able to reproduce and fix the problem. The issue is a misunderstanding on my part on how the `_thread` module works. In the Pico, you can only start a single background thread, and any more attempts return an error. I have now removed the invalid usage and Microdot will work in synchronous mode, which in my testing works well.

The concurrency options for Microdot are then as follows:

- coroutines, with the `microdot-asyncio` module and the `uasyncio` module from the MicroPython library
- threads, with the `threading` module from the MicroPython library (as long as the board/platform supports multithreading)

miguelgrinberg, 2022-08-07:
I have released version 1.0 of Microdot with many new features and improvements. Please reopen if there are still problems with this.

closed 2022-08-07 by miguelgrinberg