miguelgrinberg/microdot #50: Microdot, Micropython and Raspberry Pi Pico W

cheeseshark, 2022-08-23:
I am attempting to use Microdot with Micropython v1.19.1-88-g74e33e714 on a Raspberry Pi Pico W.

When I attempt to import a class from the microdot.py file (from microdot import Microdot) I receive the following error:

**Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'microdot.Microdot'**

When I import as a whole (import microdot) it imports, but none of the classes are available (app = Microdot() )

Is this an issue with Microdot not being supported by this version of Micropython?

 Thanks in advance

-- Mike

miguelgrinberg, 2022-08-23:
No, this has to be something that is wrong on your side, since I've personally tested the Pico W recently.

Can you show me a directory listing with all the files that you have on your device, including their sizes? My guess is that your microdot.py is corrupted or invalid.

cheeseshark, 2022-08-23:
I'm using Thonny, I'm not sure how to show details on the file list, but I can show individual file properties.

I've attached a screen shot of the Thonny interface.
[image]

I downloaded the code, and unzipped it to a directory, then uploaded the microdot.py to the device using Thonny.

raven703, 2022-08-23:
probably it's trying to import from microdot folder and not microdot.py file.

cheeseshark, 2022-08-23:
> probably it's trying to import from microdot folder and not microdot.py file.

@raven703 that was it exactly!  Thanks raven703!

miguelgrinberg, 2022-08-23:
Oh, I see, so you put the microdot files inside a `microdot` folder. Then the import should have been:

    from microdot.microdot import Microdot

The first `microdot` is for your folder, the second is for the `microdot.py` file. It's better to move the files to the top and not use a folder though.