miguelgrinberg/microdot #51: CircuitPython support vs. switching to MicroPython

sgbaird, 2022-08-27:
I built up a [basic Pico W - compatible example for an autonomous research lab demo](https://github.com/sparks-baird/self-driving-lab-demo/tree/main/scripts/pico-w) using CircuitPython libraries like [`neopixel`](https://github.com/adafruit/Adafruit_NeoPixel) and [`adafruit_as7341`](https://github.com/adafruit/Adafruit_AS7341). If I want to use microdot, seems I'll need to switch over to running MicroPython and then use the Blinka compatibility layer to access the CircuitPython libs (and hope that what I want is supported). Does that seem like the best option? Any feedback/suggestions?

miguelgrinberg, 2022-08-27:
Is there anything in this package that is incompatible with CircuitPython? The language is always MicroPython. As long as the low level socket support exists, Microdot should run fine.

sgbaird, 2022-08-27:
@miguelgrinberg thanks, turns out it's an issue of CircuitPython not supporting Pico W yet https://github.com/adafruit/circuitpython/issues/6558. So, I'll need to set it up per MicroPython's instructions.

closed 2022-08-27 by sgbaird