miguelgrinberg/microdot #52: [FEATURE REQUEST] example of controlling an LED and reading temperature from Pico W

sgbaird, 2022-08-27:
Sort of an IoT "Hello, World!" example using microdot and Pico W, similar to:
- https://www.raspberrypi.com/news/how-to-run-a-webserver-on-raspberry-pi-pico-w/
- https://github.com/recantha/PicoTemp
- https://www.tomshardware.com/how-to/send-and-receive-data-raspberry-pi-pico-w-mqtt
- https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf
- etc.

It seems like the code could be a lot more succinct using Microdot.

sgbaird, 2022-08-27:
Just noticed the [GPIO example](https://github.com/miguelgrinberg/microdot/blob/main/examples/gpio/gpio.py).

miguelgrinberg, 2022-08-27:
Yeah, I'm not sure there is a need to have examples for specific boards. There are just too many.

closed 2022-08-27 by miguelgrinberg

sgbaird, 2022-08-27:
All good! Thanks for the quick responses.