miguelgrinberg/microdot #54: Pass parameter/context to html

djs0109, 2022-08-30:
Hello,

I am looking for a way to adjust my HTML dynamically.

My use case is, that I want to develop a web UI for ESP32 MCU, where the users can configure the WLAN. Therefore, the HTML should be but not static but dynamic (based on the scanned WLAN list). Here I found an [example](https://github.com/tayfunulu/WiFiManager/blob/master/wifimgr.py), but not using microdot.

raven703, 2022-09-01:
You can use Javascript to update your html dynamically.  Browser will send data to microdot server and microdot will serve HTML pages back. For data exchange you can use either Web sockets or Server Side Events with fetch().

miguelgrinberg, 2022-09-01:
If you need to generate your HTML dynamically you can consider using templates. I have recently added support for the utemplate library. You can find a couple of [examples](https://github.com/miguelgrinberg/microdot/tree/main/examples/hello).

closed 2022-09-05 by miguelgrinberg