miguelgrinberg/microdot #56: problem with more than one connection on pyboard + w5500

Morteza-24, 2022-09-08:
I can run a simple web server ([the getting started example](https://microdot.readthedocs.io/en/latest/intro.html#getting-started)) and access it on a single device, but when I try to access it on a second device it stops working with the following error:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "microdot.py", line 926, in run
  File "microdot.py", line 35, in create_thread
  File "microdot.py", line 968, in handle_request
  File "microdot.py", line 482, in write
OSError: 7`

and the error isn't thrown right away, usually after I try to access it again from the first device that I tried it on.

This happens specially on chrome which seemingly sends a couple more packets than chromium.

I was guessing maybe it's a firmware-related problem (cause of the error code 7: 'Argument list too long') but I've got the latest micropython firmware with _MICROPY_PY_NETWORK_WIZNET5K_ option enabled. I'm running this on a pyboard v1.1 + W5500 if that matters.

thank you in advance.

miguelgrinberg, 2022-09-08:
I don't know what version of microdot.py you are using, I cannot map the line numbers to a recent version.

Morteza-24, 2022-09-08:
Sorry I forgot I had added a couple of comments to the source code. Here's the output of the latest version:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "microdot.py", line 920, in run
  File "microdot.py", line 32, in create_thread
  File "microdot.py", line 962, in handle_request
  File "microdot.py", line 479, in write
OSError: 7
```

sgbaird, 2022-09-08:
I think I ran into the same or a similar issue trying to access a web server on multiple devices, but I figured that was due to an issue on my end not explicitly handling a case with multiple accesses. Is the intended behavior that it can be accessed via multiple devices out of the box?

miguelgrinberg, 2022-09-08:
@sgbaird it's a web server, it has no knowledge of how many devices are accessing it. All it knows is that someone is sending requests, but there is nothing that changes when the requests come from different places.

@Morteza-24 I didn't think error 7 can happen during a write, I wonder if this is some weird implementation thing in the network stack of your device, which is using error 7 to mean some other condition.

Morteza-24, 2022-09-08:
Well that's a pretty weird behaviour since I can refresh the page all I want but as soon as I open it on another device it starts to act weird and soon it will throw an error. And even if it's a lower level problem related to the wiznet and/or micropython I still wonder why it whould care what device a request is coming from. Maybe it could only keep one tcp connection open? Because I see that browsers send some keep alive requests. I would be glad to get your suggestion on how to pinpoint this problem, otherwise I would just look into the wiznet and micropython a little bit more. And thank you again.

bulletmark, 2022-09-08:
FYI, you can use the `ab` (apache benchmarking tool) to blast multiple concurrent requests. I just now updated to the latest microdot versions and tested this. I don't get any issue if I hit my ESP32 with hundreds of requests with up to 6 in parallel. With 7 or more I find the client reports a socket connection error but the server reports nothing and microdot keeps running. Presumably that is just some MicroPython or platform resource constraint. It makes no difference if I run `ab` on multiple client machines, just that limit of 6 concurrent requests always applies.

So it appears to me that microdot has no issue with multiple concurrent requests from multiple machines.

sgbaird, 2022-09-09:
> @sgbaird it's a web server, it has no knowledge of how many devices are accessing it. All it knows is that someone is sending requests, but there is nothing that changes when the requests come from different places.

@miguelgrinberg thanks for clarifying. Am I missing something basic here, then? https://www.screencast.com/t/KzHazB7O

And trying again, it [seems to be working](https://www.screencast.com/t/PePdNnaZoOZ). Hmm..

[Server code](https://github.com/sparks-baird/self-driving-lab-demo/blob/main/src/micropython_sdl_demo/main.py)

> Well that's a pretty weird behaviour since I can refresh the page all I want but as soon as I open it on another device it starts to act weird and soon it will throw an error. ...

I'm seeing similar behavior to this in the first case, but not running into issues the second time. So, I'm guessing it's something specific to my laptop/chrome/etc. and not an issue with Microdot.

# First instance
[image]

# Second instance
[image]

Morteza-24, 2022-09-09:
@bulletmark thank you for the information.
What is your concurrency mode? Does micropython have threading on the esp port?

miguelgrinberg, 2022-09-09:
@sgbaird Do you see this behavior with the hello.py example from this repo? The problem could be related to a number of things, so instead of using your own application, let's simplify and try an example that has the simplest logic.

bulletmark, 2022-09-09:
I am using `microdot_asyncio`.

Morteza-24, 2022-09-10:
I get this error when I try the [hello_async.py](https://github.com/miguelgrinberg/microdot/blob/main/examples/hello/hello_async.py) example:
Starting async server on 0.0.0.0:5000...
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "microdot_asyncio.py", line 318, in run
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "microdot_asyncio.py", line 276, in start_server
  File "uasyncio/stream.py", line 1, in start_server
OSError: [Errno 107] ENOTCONN

```

sgbaird, 2022-09-10:
> @sgbaird Do you see this behavior with the hello.py example from this repo? The problem could be related to a number of things, so instead of using your own application, let's simplify and try an example that has the simplest logic.

It's not a blocker for me right now, just something I noticed. A MWE would be good. I'll try to circle back to this soon and try with `hello.py` to see if I run into the same issues.

closed 2022-09-12 by Morteza-24