miguelgrinberg/microdot #57: WiFi connection function?

simonmonk, 2022-09-09:
I'm using microdot on the Raspberry Pi Pico W and it's great to see some Pico web server code at the right level of abstraction, rather than the ugly socket stuff. 
For me, it would be really nice to have a utility function in microdot.py that does the WiFi connection (see below). I know its perhaps not generic enough, but it makes the main web server code very much cleaner. What do you think?
Sorry not to do this as a pull-request, but (to be honest) I don't really know what I'm doing with git.

Proposed new function:
```
def connect_to_network(wlan, ssid, password):
    wlan.active(True)
    wlan.config(pm = 0xa11140)  # Disable power-save mode
    wlan.connect(ssid, password)

    max_wait = 10
    while max_wait > 0:
        if wlan.status() < 0 or wlan.status() >= 3:
            break
        max_wait -= 1
        print('waiting for connection...')
        time.sleep(1)

    if wlan.status() != 3:
        raise RuntimeError('network connection failed')
    else:
        print('connected')
        status = wlan.ifconfig()
        print('ip = ' + status[0])
```

This the simplifies my server code to:

```
from microdot import Microdot, connect_to_network
import network
import time

ssid = 'Monk'
password = 'd4daaa7eda'

wlan = network.WLAN(network.STA_IF)
app = Microdot()
        
print('Connecting to Network...')
connect_to_network(wlan, ssid, password)

@app.route('/')
def index(request):
    return 'Hello, from Pico'

app.run(port=80)
```

miguelgrinberg, 2022-09-09:
Microdot is a generic web framework, it is not concerned with low-level details such as the configuration of the network layer. It would be huge effort to support all possible connection modes for all devices, and this isn't the goal of this library, so I do not currently plan on adding this.

What you should do is create a wifi.py module with your connection function, and drop it alongside microdot.py to use in your applications.

closed 2022-09-09 by miguelgrinberg

simonmonk, 2022-09-09:
You are quite right in all respects - thanks for the pointer.