miguelgrinberg/microdot #58: How to find the IP address of the server?

lesept777, 2022-09-12:
Hi, this is my first test of Microdot, using a T-Pico C3, equipped with an ESP32 C3.
I test the simple example below:
```
from microdot import Microdot
app = Microdot()

@app.route('/')
def index(request):
    return 'Hello, world!'

app.run(port=80)
```
But I don't know how to access to the server. using my smartphone and Httper, because I don't know the IP address of the ESP32.

I tried to do this:
```
import network
ap = network.WLAN(network.AP_IF)
print(ap.ifconfig())
```
which answers `('192.168.4.1', '255.255.255.0', '192.168.4.1', '0.0.0.0')` but I don't get any reply from the server when I send a request to 192.168.4.1 using Httper, both for GET and PUT requests. 
The same with
`app.run(host='192.168.4.1', port=80)`
How can I know the correct IP address?

beyonlo, 2022-09-12:
@lesept777 First your smartphone WiFi need to be connected to that Access Point created by you, on the ESP32-C3. If you did that, so your smartphone will receive from ESP32-C3 a IP from the network `192.168.4.x`, like as `192.168.4.2`. You can to go in your WiFi config and check what was your IP address received by ESP32-C3. If you done that, you need just to open a browser on your smartphone and open http://192.168.4.1, where `192.168.4.1` is the ESP32-C3 IP Access Point.

lesept777, 2022-09-12:
Thanks for your answer. You sid:

> First your smartphone WiFi need to be connected to that Access Point created by you, on the ESP32-C3.

But, when searching for available WiFi networks on my smartphone, I can see anything that looks like the Access Point.
If I run in Thonny:
```
from microdot import Microdot
app = Microdot()

@app.route('/')
def index(request):
    return 'Hello, world!'

app.run(host='192.168.4.1', port=80)
```

what should I see in the WiFi list on my smartphone?

beyonlo, 2022-09-12:
To create an `Access Point WiFi` is not a thing that `Microdot` coverage. The `Microdot` is a small web framework where is transparent for it how you are creating an `AP` or running in a `STA` mode. Your question is not about `Microdot`, but about `MicroPython`, where the `MicroPython` community is the right place. Anyway [here](https://docs.micropython.org/en/latest/esp32/quickref.html#networking) there is a official `MicroPython` `ESP32` port documentation about how do create the AP/STA mode and how it works.

lesept777, 2022-09-13:
Thank you @beyonlo.
I found [here](https://forum.micropython.org/viewtopic.php?f=18&t=7104&p=40422&hilit=AP_IF+create+access+point#p40422) that there is a mistake in the [documentation of Micropython's network](https://docs.micropython.org/en/latest/esp32/quickref.html#networking), in the second part of the first example: it should be `essid` instead of `ssid`
`ap.config(essid='ESP-AP') # set the SSID of the access point`

When I run this script:
```
import network
ap = network.WLAN(network.AP_IF)
ap.config(essid='ESP-AP') # essid instead of ssid
ap.active(True)
print(ap.ifconfig())
```
I can see the access point on my smartphone and connect to it. Then after creating the AP, if I run the Microdot script, I can open http://192.168.4.1/ and see the `Hello World!` message.

closed 2022-09-16 by lesept777