miguelgrinberg/microdot #59: Code called in parallel to the main http server

marcin-kasinski, 2022-09-12:
Hi.
Is it possible to have an http server running and in parallel some kind of cron that every so often will call my code completely unrelated to the HTTP microdot server?

Example : every few hours I would like to download small data from the Internet, which I will want to serve as someone calls my server.

Trivial example: every few hours, in parallel to the running http server, I would like to turn on and off the diode

beyonlo, 2022-09-12:
@marcin-kasinski Yes, is possible.

If you use the `Microdot` with `asyncio` support, you can just to create a new [asyncio task](https://docs.micropython.org/en/latest/library/uasyncio.html?highlight=asyncio#uasyncio.create_task) to do what do you want. 

If you use `Microdot` using `thread`, so you still can to do that, just creating another thread to do what do you want.

I prefer to run `Microdot` with `asyncio`.

marcin-kasinski, 2022-09-12:
@beyonlo 

Do you have a simple example here?

Let's say I want to run an http server and in parallel write out anything to the console every second.

beyonlo, 2022-09-12:
> Do you have a simple example here?

@marcin-kasinski in that url that I pasted for you, on the top of page has a example creating tasks (`uasyncio.create_task(blink(led1, 700))`).

> Let's say I want to run an http server and in parallel write out anything to the console every second.

Yes, as I told you, if you will use microdot asyncio, you need just to create a new task to do something at the same time that microdot runs - it is not blocking.

marcin-kasinski, 2022-09-13:
Thank you very much for the explanation.

closed 2022-09-13 by marcin-kasinski