miguelgrinberg/microdot #60: Handling special characters/utf-8 from request

WattageGuy, 2022-09-17:
According to the docs body string in HTTP response should be encoded in UTF-8 but the response cant handle special characters like "ÅÄÖ". This is the code im using:

app = Microdot()

htmldoc = '''<!DOCTYPE html>
<html>
    <head>
        <title>Microdot Example Page</title>
         <meta http-equiv="Content-Type" content="text/html;charset=UTF-8">
    </head>
    <body>
        <div>
            <h1>Microdot Example Page</h1>
            <p>Test åöä</p>
            <p><a href="/shutdown">Click to shutdown the server</a></p>
            <form action="/form" method="POST" accept-charset="utf-8">
            <label for="ssid">SSID:</label><br>
            <input type="text" id="ssid" name="ssid"><br>
            <label for="password">Password:</label><br>
            <input type="password" id="password" name="password"><br>
            <input type="submit" value="Submit">
            </form>
        </div>
    </body>
</html>
'''

@app.route('/')
def hello(request):
    return htmldoc, 200, {'Content-Type': 'text/html'}

@app.route('/form', methods=['GET', 'POST'])
def index(req):
    if req.method == 'POST':
        print(req.form.get('ssid')) #  <-- This outputs LÃ¶s if the given value is Lös
        ssid = req.form.get('ssid')
        password = req.form.get('password')
        print(ssid) #  <-- This outputs LÃ¶s if the given value is Lös
        print(password) #  <-- This outputs LÃ¶s if the given value is Lös
    return f'SSID: {ssid}, Password: {password}' #  <-- This outputs LÃ¶s if the given value is Lös

miguelgrinberg, 2022-09-17:
@WattageGuy This is actually a problem on my part. I documented how I intended this to work, but it appears I haven't implemented this fully. Fix coming soon.

miguelgrinberg, 2022-09-17:
@WattageGuy Could you please try the microblog.py in the main branch and confirm you are able to work with UTF-8 forms with it? Thanks!

commit 75725795 (2022-09-17): Charset handling in Content-Type headers (Fixes #60)
closed 2022-09-17 by miguelgrinberg