miguelgrinberg/microdot #62: Async websockets - Unicode Error when connecting from Dart/Flutter

ThinkTransit, 2022-09-18:
Hi,

I have a micropython app running microdot async websockets. It works ok from a browser however when I try to connect to the websocket from Dart/Flutter I get a unicode error.

```
Traceback (most recent call last):
  File "/lib/microdot_asyncio.py", line 328, in handle_request
  File "/lib/microdot_asyncio.py", line 70, in create
UnicodeError: 
```

Any ideas on how I can debug this further?

I am using the latest microdot code from github.

Thanks

miguelgrinberg, 2022-09-18:
MicroPython does not currently have SSL support in its uasyncio package. I often see the `UnicodeError` appear when the client is treating the server as a TLS server, while in fact it isn't.

ThinkTransit, 2022-09-18:
Thanks @miguelgrinberg that was the problem, I was using wss: instead of ws:

Great project btw

closed 2022-09-18 by ThinkTransit