miguelgrinberg/microdot #63: Question: upload file example.

beyonlo, 2022-09-22:
Hello.

I would like to send, via upload page, to the `ESP32` flash a `file.tar.gz` file (~`6MB`). Are there any example or tip how to do that?

The idea is just simple: user click on upload button on the page, and select on local computer the file, and send it.

Thank you.

miguelgrinberg, 2022-09-22:
There is no support for form-based (i.e. multipart) file uploads. The only way to send large files is with the client sending the binary payload directly in the body of the request, which the server can access via `request.stream` and read in chunks as necessary.

miguelgrinberg, 2022-09-22:
Quick example: https://github.com/miguelgrinberg/microdot/tree/main/examples/uploads

beyonlo, 2022-09-22:
@miguelgrinberg Thank you very much for the example, that will be very useful for everyone!

I tested and works fine for little files, but unfortunately not works with files with size more than 16384 bytes.

I did many tests until to discover that works until 16384 bytes. If I add 1 more byte, (16385 bytes) do not works anymore. I did a take a look on the `uploads.py,` but I have no idea what can be the problem.

Do you know what can be the problem?

Thank you very much!

miguelgrinberg, 2022-09-22:
You need to set a maximum allowed size here: https://microdot.readthedocs.io/en/latest/api.html#microdot.Request.max_content_length.

beyonlo, 2022-09-22:
@miguelgrinberg amazing! Works! I set up the `max_content_length` to `6MB` and works fine :)

Thanks for the help and example! :)

closed 2022-09-22 by beyonlo