miguelgrinberg/microdot #68: inherit from another template using utemplate

diegorebollo, 2022-10-08:
Hi, I using microdot utemplate implementation and everything works flawless but i want to use a base html template and inherit from that for others pages. I was wonder if this is doable, reading utemplate documentations it's looks possible but it a bit unclear how to do it. Any clue?

miguelgrinberg, 2022-10-08:
My understanding is that it does not support template inheritance. I have added a quick [bootstrap](https://github.com/miguelgrinberg/microdot/tree/main/examples/templates/utemplate) example that shows a way to fake inheritance through includes, in case it is helpful.

If you have any specific questions about uTemplate, please contact the package developer, I cannot provide support for this library beyond showing how I use it.

closed 2022-10-08 by miguelgrinberg

diegorebollo, 2022-10-08:
Thanks you very much, I really appreciate it.