miguelgrinberg/microdot #70: Does microdot WebSocket support Sec-WebSocket-Protocol (sub-protocols)

QGB, 2022-10-13:
https://developer.mozilla.org/en-US/docs/Web/API/WebSockets_API/Writing_WebSocket_client_applications#creating_a_websocket_object

https://github.com/aaugustin/websockets/blob/main/src/websockets/headers.py#L441

miguelgrinberg, 2022-10-13:
There is no support for subprotocols at this time.

QGB, 2022-10-16:
https://github.com/aaugustin/websockets 
```
2022-10-16:17:51:01,746 INFO     [server.py:707] server listening on 0.0.0.0:1122
2022-10-16:17:51:05,130 DEBUG    [protocol.py:253] = connection is CONNECTING
2022-10-16:17:51:05,132 DEBUG    [server.py:282] < GET /ws HTTP/1.1
2022-10-16:17:51:05,134 DEBUG    [server.py:284] < Host: 192.168.1.3:1122
2022-10-16:17:51:05,136 DEBUG    [server.py:284] < Connection: Upgrade
2022-10-16:17:51:05,137 DEBUG    [server.py:284] < Pragma: no-cache
2022-10-16:17:51:05,138 DEBUG    [server.py:284] < Cache-Control: no-cache
2022-10-16:17:51:05,139 DEBUG    [server.py:284] < User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.5112.102 Safari/537.36 Edg/104.0.1293.63
2022-10-16:17:51:05,140 DEBUG    [server.py:284] < Upgrade: websocket
2022-10-16:17:51:05,141 DEBUG    [server.py:284] < Origin: http://192.168.1.3:1122
2022-10-16:17:51:05,141 DEBUG    [server.py:284] < Sec-WebSocket-Version: 13
2022-10-16:17:51:05,142 DEBUG    [server.py:284] < Accept-Encoding: gzip, deflate
2022-10-16:17:51:05,148 DEBUG    [server.py:284] < Accept-Language: zh-CN,zh;q=0.9,en;q=0.8,en-GB;q=0.7,en-US;q=0.6,ru;q=0.5,ja;q=0.4,zh-TW;q=0.3,it;q=0.2,de;q=0.1
2022-10-16:17:51:05,149 DEBUG    [server.py:284] < Sec-WebSocket-Key: j34CZAikuFzGn6sUzkFSoA==
2022-10-16:17:51:05,150 DEBUG    [server.py:284] < Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
2022-10-16:17:51:05,151 DEBUG    [server.py:284] < Sec-WebSocket-Protocol: tty
2022-10-16:17:51:05,154 DEBUG    [server.py:303] > HTTP/1.1 101 Switching Protocols
2022-10-16:17:51:05,155 DEBUG    [server.py:305] > Upgrade: websocket
2022-10-16:17:51:05,159 DEBUG    [server.py:305] > Connection: Upgrade
2022-10-16:17:51:05,159 DEBUG    [server.py:305] > Sec-WebSocket-Accept: 5ZDVan6JVBmGnQKKH6vUiIsIAUk=
2022-10-16:17:51:05,162 DEBUG    [server.py:305] > Sec-WebSocket-Extensions: permessage-deflate; server_max_window_bits=12; client_max_window_bits=12
2022-10-16:17:51:05,163 DEBUG    [server.py:305] > Date: Sun, 16 Oct 2022 09:51:05 GMT
2022-10-16:17:51:05,164 DEBUG    [server.py:305] > Server: Python/3.7 websockets/10.3
2022-10-16:17:51:05,165 INFO     [server.py:642] connection open
2022-10-16:17:51:05,166 DEBUG    [protocol.py:342] = connection is OPEN
tty
2022-10-16:17:51:05,171 DEBUG    [protocol.py:1404] ! failing connection with code 1006
2022-10-16:17:51:05,171 DEBUG    [protocol.py:1489] = connection is CLOSED
2022-10-16:17:51:05,173 DEBUG    [protocol.py:1309] x half-closing TCP connection
```

miguelgrinberg, 2022-10-16:
I don't know what you are you showing me. As I said, the Microdot websocket implement does not support subprotocols at this time.