miguelgrinberg/microdot #71: Question: how is possible to run Microdot in two IPs?

beyonlo, 2022-10-13:
Hi @miguelgrinberg 

Is there any way to run the microdot in different IP address?  

The ESP32 is running in STA mode, getting the IP address via DHCP, and running simultaneously on AP mode, with IP address 192.168.4.1.

I would like that if anyone access the IP 192.168.4.1 on the browser, open page x.html and when access other IP address, open the page y.html.

something like this

QGB, 2022-10-13:
use `request.headers['Host']`

miguelgrinberg, 2022-10-13:
This is something that the network stack needs to deal with, not Microdot. If passing `0.0.0.0` doesn't listen on all the interfaces, then that's a problem with the network functions.

beyonlo, 2022-10-13:
> use `request.headers['Host']`

@QGB perfect! Works very well!

Thank you so much!

beyonlo, 2022-10-13:
> This is something that the network stack needs to deal with, not Microdot. If passing `0.0.0.0` doesn't listen on all the interfaces, then that's a problem with the network functions of your board, or maybe an issue in MicroPython's implementation of networking.

@miguelgrinberg I was already using `0.0.0.0` and listening in both interfaces, the `AP` mode and `STA` mode - simultaneously, but I have just the same `html` page accessing `STA` and `AP` mode. If we consider that `AP` mode is one interface and `STA` mode is another interface, so passing `0.0.0.0` listen in all the interfaces. Am I correct?

I'm using the @QGB tip and works very well. So, I just getting the `IP address` from `request.headers['Host']` and if the host is '192.168.4.1' I return `x.html` else return `y.html.` Was a perfect solution for my case.

miguelgrinberg, 2022-10-13:
@beyonlo Got it. For some reason I thought the problem was that connecting was the problem, possibly because of the wording of your question. The issue was in handling the two IPs differently more than running Microdot on the two IPs, which you already were.

beyonlo, 2022-10-13:
@miguelgrinberg yes, you are right, writing "to run Microdot in two IPs" was not the correct way to explain my question.

Thank you for your attention!

closed 2022-10-13 by beyonlo