miguelgrinberg/microdot #76: Relative imports

DiegoPomares, 2022-10-15:
Hello again!

Would it be bad to use relative imports in the core extensions?

I'm placing external libraries into a `vendor` directory, I like the idea of having both the source files and namespaces separated from the actual application, but when I added `microdot_asyncio.py` I had to make relative the `microdot` imports so that it would work.

For reference, here's the the [app repo](https://github.com/DiegoPomares/fermax_plus)

Disclosure: This is my first MicroPython project so I'm still figuring things out and I'm unaware about best practices, just doing what seems best and iterating over it.

miguelgrinberg, 2022-10-15:
Relative imports only work for files that are inside a package. Microdot is a collection of top-level modules, so it cannot be done. It only works for you because you are moving these files into a `vendor` package.

DiegoPomares, 2022-10-16:
Ok makes sense when you put it that way :+1:

closed 2022-10-16 by DiegoPomares