miguelgrinberg/microdot #81: having issues setting up a route

codiak6335, 2022-11-15:
I hope this is user error... but trying to create a route for "change" and it always 404's when called

@app.route('/change', methods=['GET', 'POST'])
def change(request):
     print (request.args)
    return send_file('Interval.html')

http://192.168.0.134:5000/change?time=11&recaptchaResponse=

miguelgrinberg, 2022-11-15:
The problem must be somewhere else, not in the code that you are showing me. What else do you have in your application? Is it a single file? Multiple files? What's the structure?

codiak6335, 2022-11-15:
Looks like it was user error and I still don't know where...  but in stripping my code to post, I actually got it to work.   Thanks for the followup and sorry for wasting your time.

closed 2022-11-15 by miguelgrinberg