miguelgrinberg/microdot #82: file upload example

mc0110, 2022-11-16:
Hi Miguel,

it's Magnus once again. Thanks for the great framework.

Your upload example works in the threat variant in my code too, but not in the asyncio variant. Do you have an idea why? The error is: TypeError: object with buffer protocol required.

I have another question. Why can't I use global variables in the code that I need to query / set between requests.

miguelgrinberg, 2022-11-16:
I cannot comment on code that I haven't seen.

But in any case, I've added an asyncio version of the file upload example: https://github.com/miguelgrinberg/microdot/blob/main/examples/uploads/uploads_async.py.

mc0110, 2022-11-16:
That is really true! 

I'll try it with your example. If I get stuck, I'll post my code example.

mc0110, 2022-11-16:
Can you give me a hint on the global variables question - it's a mystery to me

mc0110, 2022-11-16:
Is it true that the magic line 
`Request.max_content_length = 1024 * 1024 # 1MB (change as needed)`
is? Could you please explain this?

miguelgrinberg, 2022-11-16:
I can't answer your global variables questions, because I don't understand what you are asking me. Same as before, without code it is difficult. Microdot has nothing to do with global variables, you should be able to use them when it makes sense.

The `max_content_length` is the maximum size you want to allow for uploads. You don't have to ask me about this, by the way, this is all documented: https://microdot.readthedocs.io/en/latest/api.html#microdot_asyncio.Request.max_body_length

miguelgrinberg, 2022-11-17:
> After uploading I would like to jump back to the previous path, but unfortunately I end up on the start page, no matter what I specify as response. I tried "redirect, response, request_request".

What does "previous path" mean here? How would the server know what the previous path is? All you need to do after you save the file is to return a redirect to the place you want to go. The after_request thing that you are doing does not really make sense in this case.

> I unfortunately lose the cookies on "POST" returns. What possibility do you recommend to pass on data even with POST?

What cookies? You don't have any cookies in the code you copied in your question. Cookies cannot be "lost" because they are kept by the client. Nothing you can do on the server other than explicitly telling the client to delete can make a cookie disappear.