miguelgrinberg/microdot #85: serveur sending websocket message before shutdown

sophiedegran, 2022-12-05:
Miguel

All work fine on raspberry pico it's amazing project!

I manage a websocket list like this:
[token_key, ip, sock, web_socket_key]

@app.route ('/update')
@with_websocket

def update (request, ws):
token_key = random value 
ip = request.client_addr
sock = request.sock
web_socket_key = request.headers.get ('Sec-WebSocket-Key', '')

How can I send to the list the message ? I suppose i have to create a request in microdot with specifiq websocket ?
Please help me! It's really hard for me to do that !!!

Regards

miguelgrinberg, 2022-12-05:
I'm sorry but I don't understand what you want to do. You mentioned a list, but there is no list in your code. And what is this shutdown reference in the title? I think you are not giving me the full context.

sophiedegran, 2022-12-06:
Ok so sorry

So I manage a list as explain on each websocket open by  clients
_websocket_list.append [token_key, ip, sock, web_socket_key]

@app.route ('/shutdown')
def shutdown (request):
for x in range(len(_websocket_list) ):
  message = 'Serveur will shutdown in 5 sec'
  -->here would like to send ...

miguelgrinberg, 2022-12-06:
Something like this maybe?

```python
for ws in _websocket_list:
    ws[2].send('Serveur will shutdown in 5 sec')
```

Here I'm using the 3rd element in each of your clients, which is the `ws` object that was passed to each client handler.

miguelgrinberg, 2022-12-10:
> 1. can't explain why string (i try json data also) not accepted

I think you have not included the complete error from MicroPython, with the full stack trace. That makes it impossible for me to understand what you get the error.

> 2- remenber the page is doing on /update websocket and i send from /shutdown perharps a security reason for that point

Do you have any antivirus? Looks like this error could be related to aggressive protocol filtering. Try disabling to see if the error goes away. Reference: https://forum.eset.com/topic/8355-protocol-filtering-breaks-websockets-after-update-at-5052016/

sophiedegran, 2022-12-10:
1-
Traceback (most recent call last):
File "C:\sophie\venv\lib\site-packages\microdot.py", line 1113, in
dispatch_request
   res = f (req, **req.url_args)
File "C:\sophie\microdot\microdot_socket.py", line 638, in shutdown
  ws[2].send ('test send')
TypeError: a bytes-like object is required, not 'str'
Get /shutdown 500
Get /update 200

(for 2 clients ...)

2- will look at that thanks and tell you if improvement

Regards

sophiedegran, 2022-12-13:
Yes all work perfectly !
I was doing that because don't understand how to use ws,  now it's very clear. I have only to have a simple list with [station_A, ws_station] .
It 's a real improvement for me as we can do micro chat, interactive draw, etc.. ☺
Many many Thanks