miguelgrinberg/microdot #87: Support for Server sent events?

donskytech, 2022-12-09:
Hi,
First, I would like to say that this is really a very cool library/package.   I was able to create my [MicroPython Robot Car Using MicroDot](https://www.youtube.com/watch?v=jKnA_lBr3S8) fully controlled by your mobile phone by using the MicroDot WebSocket extension.

Is there a possibility of supporting Server Sent events in the future?  This is a really useful feature, especially in IoT projects where only push messages are needed from the web server.

Thank you for coming up with this project.

miguelgrinberg, 2022-12-09:
Nice project!

I'll keep this in mind. Or if you'd like to implement it yourself and send a PR, happy to review it and merge it.

cjogo, 2023-04-14:
Any movement on this? I have a project where CSS,HTML and JavaScript files are defined.

miguelgrinberg, 2023-12-23:
SSE is supported in Microdot 2.

closed 2023-12-23 by miguelgrinberg

donskytech, 2023-12-24:
> SSE is supported in Microdot 2.

Yes, this is quite exciting!  I can't wait to test it and write interesting blog content about version 2 of MicroDot.
I am just waiting for examples (or if they are still being worked on) before writing about this awesome and minimalistic framework.

miguelgrinberg, 2023-12-24:
@donskytech There is a simple example [here](https://github.com/miguelgrinberg/microdot/blob/main/examples/sse/counter.py).