miguelgrinberg/microdot #97: How to set headers for all responses (and errors)

damiencorpataux, 2023-02-06:
Hello,

My project need all responses to have the header `Access-Control-Allow-Origin: *`.

Using `@after_request` works well for `2xx` responses and `@errorhandler(Exception)` for `Exception`s. But this does not apply to `HTTPError`s (mainly 400, 404, 405, 500 etc).

Is there a way to set a single handler for any `HTTPError`, or better: a handler that is ensured to run on every response generated by microdot ?

miguelgrinberg, 2023-02-06:
Okay, this is something I have no considered. Basically you want a catch-all handler for all the 4xx/5xx error codes. I'll look into it.

damiencorpataux, 2023-02-06:
Here are thoughts:

- Apply `@after_request` handlers to all responses ([example](https://github.com/damiencorpataux/microdot/tree/dev-afterrequest-always)) (i believe [flask does that](https://flask.palletsprojects.com/en/2.2.x/api/#flask.Flask.after_request))

- Allow `@errorhandler(0)` to apply the handler as fallback to all status codes of `HTTPException`, if a more specific handler does not exist ([example](https://github.com/damiencorpataux/microdot/blob/dev-errorhandler-catchall/src/microdot.py#L1096))

miguelgrinberg, 2023-02-06:
Yes, I think the first option is what makes the most sense. The handler receives the response object, so it can check its attributes if it needs to know if this is a valid or error response. And for CORS support where it doesn't really matter you can just add the header(s) and forget about it.

miguelgrinberg, 2023-02-06:
After seeing how your PR caused some tests to fail I realized this change might also affect existing applications, so instead I have added a separate `@after_error_request` decorator. Let me know what you think.

commit deede1ff (2023-02-06): Add @after_error_handler decorator (Fixes #97)
commit fcaeee69 (2023-02-06): Add @after_error_handler decorator (Fixes #97)
closed 2023-02-06 by miguelgrinberg