miguelgrinberg/microdot #103: WebSocket client to external server

donskytech, 2023-02-20:
Hey,
Do you think I can use MicroDot to connect to an external WebSocket server?
I know MicroDot has a decorator for WebSocket and I have used that already.

My use case is that I wanted to initiate a WebSocket connection from the MicroDot app, not the browser.
Thanks.

miguelgrinberg, 2023-02-20:
No. Microdot has a WebSocket server. You need a WebSocket client to initiate a connection to another server.

donskytech, 2023-02-20:
Hey, thanks for the quick response.
Do you know what WebSocket client is compatible with the MicroDot app? Or could you suggest a WebSocket client that will work best with the MicroPython application running MicroDot?

miguelgrinberg, 2023-02-20:
I'm honestly not aware of what WebSocket clients for MicroPython exist, as I have been only concentrating on the server side of the protocol.

donskytech, 2023-02-20:
I will keep looking but thanks for the reply.

closed 2023-02-20 by donskytech