miguelgrinberg/microdot #104: Question: Serving files with asyncio module

ThinkTransit, 2023-02-21:
Hi,

Great library! This isn't an issue but a question if you can please help.

Is there any special configuration required for serving files when using asyncio?

I have tried the following but I'm getting a buffer TypeError when accessing /test.txt

```python
from microdot_asyncio import Microdot, Response
from microdot_utemplate import render_template
from microdot import send_file

app = Microdot()
Response.default_content_type = 'text/html'

@app.get('/test.txt')
async def index(request):
    return send_file('/static/test.txt')
```  

```python
Starting async server on 0.0.0.0:80...
Task exception wasn't retrieved
future: <Task> coro= <generator object 'serve' at 2000f4d0>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "microdot_asyncio.py", line 268, in serve
  File "microdot_asyncio.py", line 336, in handle_request
  File "microdot_asyncio.py", line 157, in write
  File "uasyncio/stream.py", line 1, in stream_awrite
  File "uasyncio/stream.py", line 1, in write
TypeError: object with buffer protocol required
```

Any ideas?

Thanks

miguelgrinberg, 2023-02-21:
Your import is incorrect. Try:

```python
from microdot_asyncio import send_file
```

ThinkTransit, 2023-02-22:
Perfect thank you!

closed 2023-02-22 by ThinkTransit