miguelgrinberg/microdot #128: Create a "mini" Microdot with reduced features

miguelgrinberg, 2023-04-11:
While Microdot fits in most microcontroller architectures, in some such as the ESP8266 it is a tight fit. It would be nice to have a completely stripped-down version that can be used for a very basic server.

rkompass, 2023-06-16:
Perhaps it matters in this context:
I observed that the nice data structures `NoCaseDict` and `MultiDict` are both used just in one place, at least in `microdot.py` itself.
I did an attempt to get rid of these but apparently missed something in recoding (website now renders without formatting).
The memory consumption reduction by that was about 2K.
@miguelgrinberg If you are interested I can post my changes.

Thank you for this nice project and thanks for this self-assignment.
What I would appreciate most was a compact (1 file) asynchronous version of microdot. Something like tinyweb but in the better microdot style and working (tinyweb is not kept up to date).
I'm interested in controlling robots over Wifi that have some activity of their own. So an async server is needed.

miguelgrinberg, 2023-06-16:
@rkompass this is all coming in version 2. My current plan (may still change) is to have four "starter" scripts: the regular and async versions, plus two "mini" versions with only the most basic support. These are all going to be single file modules.

> So an async server is needed.

Some boards offer limited multi-threading support. For example the Raspberry Pi Pico and Pico W have two cores, and can run two threads concurrently.

miguelgrinberg, 2023-12-23:
Microdot 2 is now released. I ended up not implementing what I describe above, and instead opted for a simpler solution that reduces the size of the code by making everything work asynchronously (i.e. the sync support code has been removed). The documentation also includes information on cross-compiling to mpy and freezing the modules to save space.

closed 2023-12-23 by miguelgrinberg