miguelgrinberg/microdot #129: WebSocket error (and disconnect ws client) when connected via AP and STA mode.

beyonlo, 2023-04-13:
I'm using `ESP32-S3` with `Micropython 1.19.1` with `Microdot` that I downloaded today. Older versions of `Microdot` has the same error.

**Describe the bug**

```Python
Traceback (most recent call last):
  File "microdot_asyncio.py", line 370, in dispatch_request
  File "microdot_asyncio.py", line 443, in _invoke_handler
  File "microdot_asyncio_websocket.py", line 97, in wrapper
  File "<stdin>", line 191, in ws_server
  File "microdot_asyncio_websocket.py", line 20, in receive
  File "microdot_asyncio_websocket.py", line 26, in send
  File "microdot_websocket.py", line 98, in _encode_websocket_frame
TypeError: unsupported types for __or__: 'int', 'bytes'
```

**To Reproduce**
Steps to reproduce the behavior:

Unfortunately I can't found exactly how to reproduce, because do not happen every time, but that happen just in moments that I'm using `WebSocket` connected via WiFi AP mode and WiFi Client mode. I mean, I'm using `WebSocket` connections simultaneously via IP 192.168.4.1 `(AP)` and via IP 192.168.43.143 `(STA)`

miguelgrinberg, 2023-04-14:
Is it relatively easy for you to reproduce this error? In that case, can I ask you to add some print statements that can help me understand what's going on here?

Right above [line 20](https://github.com/miguelgrinberg/microdot/blob/main/src/microdot_asyncio_websocket.py#L20) of microdot_asyncio_websocket.py, add a print statement as follows:

```python
    async def receive(self):
        while True:
            opcode, payload = await self._read_frame()
            send_opcode, data = self._process_websocket_frame(opcode, payload)
            if send_opcode:  # pragma: no cover
                print('opcode', opcode)  # <-- add this and the follow 3 prints
                print('payload', payload)
                print('send_opcode', send_opcode)
                print('data', data)
                await self.send(send_opcode, data)
            elif data:  # pragma: no branch
                return data
```

Then once you get the error, the stack trace should be preceded with four debugging lines that I'd like to see.

beyonlo, 2023-04-14:
Hello @miguelgrinberg 

Follow the error with the prints as you requested:

```Python
opcode 9
payload b'PING'
send_opcode 10
data b'PING'
Traceback (most recent call last):
  File "microdot_asyncio.py", line 370, in dispatch_request
  File "microdot_asyncio.py", line 443, in _invoke_handler
  File "microdot_asyncio_websocket.py", line 101, in wrapper
  File "<stdin>", line 191, in ws_server
  File "microdot_asyncio_websocket.py", line 24, in receive
  File "microdot_asyncio_websocket.py", line 30, in send
  File "microdot_websocket.py", line 98, in _encode_websocket_frame
TypeError: unsupported types for __or__: 'int', 'bytes'
```

Thank you

beyonlo, 2023-04-14:
@miguelgrinberg Just a fix information that I told you: that error do not happen just when are using ws simultaneously in AP and STA mode. That error happen when using just via AP too.

closed 2023-04-14 by miguelgrinberg

beyonlo, 2023-04-14:
@miguelgrinberg I updated the `microdot_websocket.py` and `microdot_asyncio_websocket.py` and until now I have no erros anymore. :partying_face: 

Thank you very much!

commit 87cd098f (2023-04-14): WebSocket error when handling PING packet (Fixes #129)