miguelgrinberg/microdot #132: Crashing when using öäü special chars in html input as form

maxi07, 2023-05-10:
**Describe the bug**
I am running a sample app to catch user input from a textbox. The html displays a textbox and a submit button. When the button is pressed, the textbox content is printed to the console just fine. If the textbox contains special characters like _äöü_, the `urldecode_bytes` function crashes with a **UnicodeError**.

**To Reproduce**
```HTML
<!DOCTYPE html>
<html>
    <head>
        <title>Microdot Example Page</title>
    </head>
    <body>
        <div>
            <h1>Microdot Example Page</h1>
            <p>Hello from Microdot!</p>
            <p><a href="/shutdown">Click to shutdown the server</a></p>
            <form action="/" method="post">
                <input name="test-input" id="test-input"/>
                <button name="test-button" id="test-button" type="submit">Submit</button>
            </form>
        </div>
    </body>
</html>
```

```Python
@app.post('/')
async def hello2(request):
    print(request.form['test-input'])
    return htmldoc, 200, {'Content-Type': 'text/html'}
```

**Expected behavior**
I expect microdot to also decode special characters.

**Additional context**
You are doing amazing work!

miguelgrinberg, 2023-05-10:
What browser is this? Internet Explorer by any chance?

maxi07, 2023-05-11:
Haha, no - I am going with Microsoft Edge (Windows 11, Version 113.0.1774.35) and Safari for iOS 16.5.
If the textbox contains "text", it works fine, as `urlencoded` is `b'test-input=test&test-button='`.
If the textbox input is "test ä", the program crashes with the given error. `urlencoded` is then equal to `b'test-input=test+%E4&test-button='`

miguelgrinberg, 2023-05-12:
@maxi07 Okay, here is the solution:

```html
<!DOCTYPE html>
<html>
    <head>
        <title>Microdot Example Page</title>
        <meta charset="UTF-8">          <--! add this line -->
    </head>
    <body>
        <div>
            <h1>Microdot Example Page</h1>
            <p>Hello from Microdot!</p>
            <p><a href="/shutdown">Click to shutdown the server</a></p>
            <form action="/" method="post">
                <input name="test-input" id="test-input"/>
                <button name="test-button" id="test-button" type="submit">Submit</button>
            </form>
        </div>
    </body>
</html>
```

This tells the browser that the page and any form submissions originating from it should be encoded in UTF-8.

commit f81de6d9 (2023-05-12): Explicitly set UTF-8 encoding for HTML in examples (Fixes #132)
closed 2023-05-12 by miguelgrinberg