miguelgrinberg/microdot #137: Int pattern does not handle negative numbers

dannystaple, 2023-05-21:
**IMPORTANT**: If you have a question, or you are not sure if you have found a bug in this package, then you are in the wrong place. Hit back in your web browser, and then open a GitHub Discussion instead. Likewise, if you are unable to provide the information requested below, open a discussion to troubleshoot your issue.

**Describe the bug**
A negative number is not matched by the int type in a url pattern.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a url pattern like `@app.route('/set_servo/<int:position>`
2. Send a value like `/set/servo/-30` (to move the servo 30 degrees anti clockwise from neutral)
3. You will receive a 404 as the pattern will not be matched.

**Expected behavior**
The function attached to the url pattern would receive the argument position=-30.

**Additional context**
Note that the unit tests only verify positive integer numbers too.

miguelgrinberg, 2023-05-21:
Oh, wow, what an oversight. I'll take care of this!

closed 2023-05-21 by miguelgrinberg

miguelgrinberg, 2023-05-21:
The fix is in release 1.3.1.

commit a0dd7c8a (2023-05-21): Support negative numbers for int path components (Fixes #137)