miguelgrinberg/microdot #148: One ESP32: Out of memory with quite a basic setup?

hagenbuch, 2023-06-15:
**Describe the bug**

I've installed micropython 1.20 on an Olimex ESP32-POE device. It runs fine. After I established an LAN connection, I just try to import microdot with the following result:

    ----- main.py -----
    >> Trying to import cfg.py: Success
    >> Trying to mount SDCard on /flash: Success
    machine_id: 240ac49ef998
    cpu_freq: 160.0
    flash_size: 0x400000
    flash_user_start: 0x200000
    >> Trying to bring up LAN:
    Waiting for IP: . 192.168.13.57
    Started webrepl in normal mode
    >> Fetching unix time from network: 20230615 12:43:59
    Traceback (most recent call last):
    File "main.py", line 437, in <module>
    MemoryError: memory allocation failed, allocating 2344 bytes
    MicroPython v1.20.0 on 2023-04-26; Olimex ESP32 ETH with ESP32

**To Reproduce**

My code looks like this:

    # -------- Webserver --------
    #import uasyncio
    from microdot import Microdot
    app = Microdot()
    @app.route('/')
    def index(request):
        return webpage(cfg)
    app.run()

**Expected behavior**

I wouldn't expect an error just by importing the library..

miguelgrinberg, 2023-06-15:
Is there any chance any of the 436 lines that precede your microdot import are consuming a large part of your memory? You are calling this a "basic setup", but my definition of basic setup would be that you run the web server on its own. These devices have limited resources. If you don't pre-compile your sources, then you must import the largest files first, so that there is more memory available to the compiler. Or else, you should seriously consider pre-compiling, or better yet, freezing as much of your code and libraries as possible.

miguelgrinberg, 2023-06-15:
@hagenbuch Most MicroPython ports do not have threading. It is expect for that import to fail. If you are getting any errors, you have to show me the statements that you issued and the errors if you need me to comment on them.

damiencorpataux, 2023-06-24:
@hagenbuch I also have use cases were I *really* need a web server that is *really* lightweight and simple - so that it is easy to debug and flyweight in memory. For these usage there is `micropython-aioweb`:
- Flask-like routing
- Supports HTTP, Websocket and Server-Events
- Lib size: 215 lines of code
The original repository seems stall and I am using it actively in various projects so I created an updated fork, compatible with micropython: https://github.com/damiencorpataux/micropython-uaioweb

miguelgrinberg, 2023-06-24:
@damiencorpataux Please stop advertising competing projects here. Thank you.

damiencorpataux, 2023-06-25:
Ok if you don't like helping others. That's a very disappointing attitude on an open-source plateform. Good luck with your ethics !

miguelgrinberg, 2023-12-23:
See https://microdot.readthedocs.io/en/latest/freezing.html for information on how to cross-compile and/or freeze your application and dependencies to reduce the memory footprint.

closed 2023-12-23 by miguelgrinberg