miguelgrinberg/microdot #152: UTF-8 decode error

xuancong84, 2023-06-23:
Very often, I encounter the following print stack while the HTTP server is still working fine:
```
#22 ets_task(4020f57c, 29, 3fffad20, 10)
('192.168.4.1', '255.255.255.0', '192.168.4.1', '192.168.4.1')
Traceback (most recent call last):
  File "microdot.py", line 1171, in handle_request
  File "microdot.py", line 384, in create
UnicodeError:
```
Do you know that `bytes.decode()` can fail?

miguelgrinberg, 2023-06-23:
The failure in that line would occur if the request URL is not encoded in UTF-8 characters. What is the URL the client is requesting in this case?

AlexanderKrampe, 2023-06-25:
I encontered a similar problem in asyncio

Traceback (most recent call last):
  File "microdot_asyncio.py", line 331, in handle_request
  File "microdot_asyncio.py", line 72, in create
UnicodeError: 

The requested URL was: 
https://192.168.178.65:5000 

My index.html also contains  meta charset="UTF-8". I might want to add my Browser (Firefox) pops up with an error while connecting (PR_CONNECT_RESET_ERROR)

acamu, 2023-11-10:
Hello,

I have the same thing
In my personnal case this URL:
https://192.168.1.76:80/

Regards

miguelgrinberg, 2023-11-11:
Correct, and https URL would not be a valid UTF-8 stream and would cause decode errors such as the one reported.

closed 2023-11-11 by miguelgrinberg

lucbenoit01750, 2023-12-13:
> Correct, an https URL would not be a valid UTF-8 stream and would cause decode errors such as the one reported. I will close, assuming this is what happened to you, OP. If you need more assistance, please reopen.

Hello, is there a special encoding for https url ?  Thanks.

miguelgrinberg, 2023-12-14:
No. Is you want to use https you have to configure the server for it. See the `ssl` option that you pass you the`run()` method.